khoih-prog/ESPAsync_WiFiManager #2: Compile Error on Platformio

Gwenselah, 2020-09-02:
Hello,

I try to compile the Async_ESP_FSWebServer example. On compiling I get this error:
[image]

My platformio.ini looks like:
>[env:wemos_d1_mini32]
>platform = espressif8266
>board = d1_mini
>board_build.filesystem = littlefs
>framework = arduino
>monitor_speed = 115200
>lib_deps =
>  ESP Async WebServer
>  ESPAsyncTCP

What I am missing?

kind regards

khoih-prog, 2020-09-03:
@Gwenselah 

Thanks for using the library.

You have to update to use the new version [**ESPAsync_WiFiManager v1.1.1**](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.1.1) to support MultiiFi

Check at then update either in 

1. **Platform.io** => [ESPAsync_WifiManager](https://platformio.org/lib/show/11065/ESPAsync_WifiManager)
2. **GitHub** => [ESPAsync_WiFiManager v1.1.1](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.1.1)

closed 2020-09-03 by khoih-prog

kaspaas, 2020-09-03:
Hi,

I'm getting the same error 

[image]

All on rev 1.1.1

[image]

khoih-prog, 2020-09-03:
@Gwenselah 

If you still have the same error, try adding the forward declaration as follows:

```cpp
...
String separatorLine = "===============================================================";

uint8_t connectMultiWiFi(void);

//format bytes
String formatBytes(size_t bytes) 
...
```

Gwenselah, 2020-09-03:
@khoih-prog This additional declaration helped to compile the code.
Thank you for your assistance

khoih-prog, 2020-09-17:
HI @Gwenselah , @kaspaas 

Just let you know the new [**ESPAsync_WiFiManager release v1.1.2**](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.1.2) already fixes the issue as well as some bugs in examples.