khoih-prog/ESPAsync_WiFiManager #3: Hi,

kaspaas, 2020-09-03:
Hi,

I'm getting the same error 

[image]

All on rev 1.1.1

[image]

_Originally posted by @kaspaas in https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/2#issuecomment-686492864_

khoih-prog, 2020-09-03:
Hi @kaspaas 

Can you try adding the forward declaration as follows:

```cpp
...
String separatorLine = "===============================================================";

uint8_t connectMultiWiFi(void);

//format bytes
String formatBytes(size_t bytes) 
...
```

closed 2020-09-03 by khoih-prog

kaspaas, 2020-09-04:
Works perfect!

Thanks!