khoih-prog/ESPAsync_WiFiManager #4: load config file issue with ESP8266 (LittleFS and SPIFFS)

rorosaurus, 2020-09-08:
Hello, I'm using [this board](https://smile.amazon.com/Makerfocus-Motherboard-Integrated-ESP8266-Compatible/dp/B075H8X7H2/), which I can't perfectly identify. I believe the flash size is 1-2MB, based on the Sketch size %, and since using 4MB flash partitions results in failure to begin either format.
esptool.py says `Auto-detected Flash size: 2MB`.

When I run the example code from Async_ConfigOnSwitch, I see a few things. I've manually censored SSID and password in this serial output:
````
Stored: SSID = ssid , Pass = password
[WM] * Add SSID =  ssid , PW =  password
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] failed
````

So it appears that the last known working configuration is recovered from the esp8266 flash. (As detailed [here](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#flash-layout))
The file system begins correctly, but `FileFS.open(CONFIG_FILENAME, "r")` returns `false`.

Then, I run the config portal and input the wifi info! It saves correctly!

````
[WM] SaveWiFiCfgFile 
[WM] OK
````

However, if I reset the device, I get the same input for `LoadWifiCfgFile` as before. `file` is false, and the load fails. I can't figure out why this is.

I wrote [this separate small sketch](https://gist.github.com/rorosaurus/4d9453473ddfa2228b4bf7ed557568db) which simply reads the config file. If I flash this sketch, instead of resetting the `Async_ConfigOnSwitch` sketch, it properly loads the file and prints it with no issue.

So it appears to only run into this problem only occurs somewhere during `Async_ConfigOnSwitch` when starting, strangely. It reproduces on both SPIFFS and LittleFS. I'm not entirely convinced it's not my hardware, to be honest.

khoih-prog, 2020-09-08:
@rorosaurus 

Thanks for using the library.

That's happening either because 

- you haven't entered ConfigPortal to input a valid sets of WiFi Credentials which would then be saved into `SaveWiFiCfgFile ` or 
- the file has been erased / corrupted between resets (forced **format**, bad flash,bad power, etc.). **I'm afraid only you can deal with this issue.**

Pls see the sequence below about how the process is working (actually just tested on a **Wemos D1**)

1. No Config Portal (CP) access and/or without valid input WiFi Credentials. This is what you get at the beginning

```
Starting Async_ConfigOnSwitch using LittleFS on ESP8266_NODEMCU
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] failed                                           <=============== always failed
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -43
[WM] Channel: 2 ,IP address: 192.168.2.186
After waiting 3.44 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH
```
---

2. Enter CP and input valid WiFi Credentials  ==>  

```
Starting Async_ConfigOnSwitch using LittleFS on ESP8266_NODEMCU
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] failed                                   <============= still failed at the beginning
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -37
[WM] Channel: 2 ,IP address: 192.168.2.186
After waiting 3.45 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHH
Configuration portal requested.
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Opening configuration portal. Got stored Credentials. Timeout 120s
[WM] WiFi.waitForConnectResult Done
[WM] SET AP_STA
[WM] 
Configuring AP SSID = ESP_705385
[WM] AP PWD = your_password
[WM] AP Channel = 11
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Custom STA IP/GW/Subnet
[WM] DNS1 and DNS2 set
[WM] setWifiStaticIP IP = 192.168.2.186
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 3.21
[WM] Local ip = 192.168.2.186
[WM] Connection result:  WL_CONNECTED
connected...yeey :)
Local IP: 192.168.2.186
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
[WM] SaveWiFiCfgFile 
[WM] OK                                        <============= after this, the file is present
[WM] freeing allocated params!
```
---

3. Reset many times => `LoadWiFiCfgFile` still OK

```
Starting Async_ConfigOnSwitch using LittleFS on ESP8266_NODEMCU
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK                                                        <========= file is still OK here
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -49
[WM] Channel: 2 ,IP address: 192.168.2.186
After waiting 3.46 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
HHHHHHH
```
---

4. Check if you accidentally turned on auto format in every startup

```
#define FORMAT_FILESYSTEM       true
```
in the code

```
// From v1.1.1
// You only need to format the filesystem once
//#define FORMAT_FILESYSTEM       true
#define FORMAT_FILESYSTEM         false

...

  if (FORMAT_FILESYSTEM) 
    FileFS.format();

  // Format FileFS if not yet
#ifdef ESP32
  if (!FileFS.begin(true))
#else
  if (!FileFS.begin())
#endif  
  {
    Serial.print(FS_Name);
    Serial.println(F(" failed! AutoFormatting."));
    
#ifdef ESP8266
    FileFS.format();
#endif
  }

```

If not this case, 

1. try on **other** boards or types of boards
2. add more debugging messages to see where it's breaking.

khoih-prog, 2020-09-08:
Try on these boards and still OK

1. **ESP8266_WEMOS_D1R1**

```
Starting Async_ConfigOnSwitch using LittleFS on ESP8266_WEMOS_D1R1
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet2, Pass = 12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet2 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -49
[WM] Channel: 2 ,IP address: 192.168.2.186
After waiting 3.47 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
```
---
2. **ESP8266_WEMOS_D1MINI**

```
Starting Async_ConfigOnSwitch using LittleFS on ESP8266_WEMOS_D1MINI
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet2 ,RSSI= -57
[WM] Channel: 4 ,IP address: 192.168.2.186
After waiting 3.56 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
[WM] freeing allocated params!
HHHHH
```

rorosaurus, 2020-09-08:
Thanks for the quick reply! And thanks for the very helpful library!

I spent a lot of time yesterday trying to figure out this issue, and I think I added a print statement inbetween some #ifdef's that didn't behave like I expected! I think the end result was `FileFS.format()` was getting called every time, even if I set the flag to false! I really should have just taken a break before looking at it again.

Regardless, I can confirm there's no issue with the library. I'm sorry to have consumed your cycles on this! Thank you again for your time and patience!

closed 2020-09-08 by rorosaurus

khoih-prog, 2020-09-08:
Good to know it's not the library's bug.

> I really should have just taken a break before looking at it again.

We all need break some time, and lot of coffee too ;-)