khoih-prog/ESPAsync_WiFiManager #8: Single SSID instead of multiple SSIDs

geosys123, 2020-09-14:
Is it possible to show only single SSID on config portal instead of SSID1 and SSID2 on config portal? Is there any to disable it or we need to make change in src files?

khoih-prog, 2020-09-14:
Hi @geosys123 

Thanks for using this new [**ESPAsync_WiFiManager Library**](https://github.com/khoih-prog/ESPAsync_WiFiManager).

Currently, it's not designed to have the single SSID feature configurable from sketch. If there are mrde demands, I'll consider to add the feature in future releases.

You can either

1. Use the previous version [**ESPAsync_WiFiManager v1.0.11**](https://github.com/khoih-prog/ESPAsync_WiFiManager#releases-1011), which uses only single SSID without the MultiWiFi feature.
2. Just **ignore** the extra SSID or make them duplicated.
3. **Change in the src files of newer Releases v1.1.1+**

closed 2020-09-14 by khoih-prog

geosys123, 2020-09-15:
Thanks for your quick reply. I may go with v1.0.11. But is there any major bug in v1.0.11 besides single SSID feature which I should  take care else I will go with newer version and modify source files.

khoih-prog, 2020-09-15:
No known bug in v1.0.11. Just single SSID feature.

You can read the releases history in its twin library

[ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager)

geosys123, 2020-09-15:
Thanks for your quick reply.

Mark-81, 2022-11-21:
Sorry to bump this thread. I'm interested too in single credential pairs. After 2 years it seems there's no way to configure the numbers of credentials from the user sketch. I think it should be enough to make the inputs `hidden` if there is a define for a single credential pairs:

```
#if SINGLE_CREDENTIALS
  #if DISPLAY_STORED_CREDENTIALS_IN_CP
  const char WM_HTTP_FORM_START[] PROGMEM = "<form method='get' action='wifisave'><fieldset><div><label>SSID</label><input value='[[ssid]]' id='s' name='s' length=32 placeholder='SSID'><div></div></div><div><label>Password</label><input value='[[pwd]]' id='p' name='p' length=64 placeholder='password'><div></div></div><input type='hidden' value='[[ssid1]]' id='s1' name='s1' length=32 placeholder='SSID1'></div></div><input type='hidden' value='[[pwd1]]' id='p1' name='p1' length=64 placeholder='password1'></div></div></fieldset>";
  #else
  const char WM_HTTP_FORM_START[] PROGMEM = "<form method='get' action='wifisave'><fieldset><div><label>SSID</label><input id='s' name='s' length=32 placeholder='SSID'><div></div></div><div><label>Password</label><input id='p' name='p' length=64 placeholder='password'><div></div></div><input type='hidden' id='s1' name='s1' length=32 placeholder='SSID1'></div></div><input type='hidden' id='p1' name='p1' length=64 placeholder='password1'><div></div></div></fieldset>";
  #endif
#else
#if DISPLAY_STORED_CREDENTIALS_IN_CP
  const char WM_HTTP_FORM_START[] PROGMEM = "<form method='get' action='wifisave'><fieldset><div><label>SSID</label><input value='[[ssid]]' id='s' name='s' length=32 placeholder='SSID'><div></div></div><div><label>Password</label><input value='[[pwd]]' id='p' name='p' length=64 placeholder='password'><div></div></div><div><label>SSID1</label><input value='[[ssid1]]' id='s1' name='s1' length=32 placeholder='SSID1'><div></div></div><div><label>Password</label><input value='[[pwd1]]' id='p1' name='p1' length=64 placeholder='password1'><div></div></div></fieldset>";
  #else
  const char WM_HTTP_FORM_START[] PROGMEM = "<form method='get' action='wifisave'><fieldset><div><label>SSID</label><input id='s' name='s' length=32 placeholder='SSID'><div></div></div><div><label>Password</label><input id='p' name='p' length=64 placeholder='password'><div></div></div><div><label>SSID1</label><input id='s1' name='s1' length=32 placeholder='SSID1'><div></div></div><div><label>Password</label><input id='p1' name='p1' length=64 placeholder='password1'><div></div></div></fieldset>";
  #endif
#endif

If you think this is worth to embed in the next commit I'll be glad to find it in the library, instead of hacking it every time I download it :-)
```

khoih-prog, 2022-11-21:
This is so easy way to read and do 

> 2. Just ignore the extra SSID or make them duplicated.

I have no time to waste and interest in going backward.

Mark-81, 2022-11-21:
Thank, very kind. Got it.