khoih-prog/ESPAsync_WiFiManager #9: connectMultiWifi not declared in this scope...

tpalaz, 2020-09-16:
Good afternoon.

Please check the below log to see full error. None of the examples in this library work. 
`Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP_WifiManager> 1.1.1
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsync_WifiManager> 1.1.1
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
src\main.cpp:182:4: warning: #warning Using static IP [-Wcpp]
   #warning Using static IP
    ^
In file included from .pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager.h:592:0,
                 from src\main.cpp:199:
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h:992:3: warning: multi-line comment [-Wcomment]
   // Add option if didn't input/update SSID/PW => Use the previous saved Credentials. \
   ^
In file included from src\main.cpp:82:0:
.pio\libdeps\esp12e\ESPAsyncWebServer-esphome\src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated (declared at C:\Users\Tyler\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
                                                                                                     ^
In file included from .pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager.h:592:0,
                 from src\main.cpp:199:
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h: In member function 'void ESPAsync_WiFiManager::scan()':
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h:557:16: warning: unused variable 'res' [-Wunused-variable]
           bool res=WiFi.getNetworkInfo(i, wifiSSIDs[i].SSID, wifiSSIDs[i].encryptionType, wifiSSIDs[i].RSSI, wifiSSIDs[i].BSSID, wifiSSIDs[i].channel, wifiSSIDs[i].isHidden);
                ^
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h: In member function 'void ESPAsync_WiFiManager::criticalLoop()':
src\main.cpp:48:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
 #define TIME_BETWEEN_MODELESS_SCANS       30000
                                           ^
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h:668:47: note: in expansion of macro 'TIME_BETWEEN_MODELESS_SCANS'
     if (scannow == -1 || millis() > scannow + TIME_BETWEEN_MODELESS_SCANS)
                                               ^
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h: In member function 'boolean ESPAsync_WiFiManager::startConfigPortal(const char*, const char*)':
src\main.cpp:45:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
 #define TIME_BETWEEN_MODAL_SCANS          20000
                                           ^
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h:798:48: note: in expansion of macro 'TIME_BETWEEN_MODAL_SCANS'
     if ( scannow == -1 || millis() > scannow + TIME_BETWEEN_MODAL_SCANS)
                                                ^
In file included from .pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager.h:592:0,
                 from src\main.cpp:199:
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h: In member function 'boolean ESPAsync_WiFiManager::isIp(String)':
.pio\libdeps\esp12e\ESPAsync_WifiManager\src/ESPAsync_WiFiManager-Impl.h:2164:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < str.length(); i++)
                                  ^
src\main.cpp: In function 'void check_WiFi()':
src\main.cpp:267:22: error: 'connectMultiWiFi' was not declared in this scope
     connectMultiWiFi();
                      ^
*** [.pio\build\esp12e\src\main.cpp.o] Error 1`

Removing the "connectMultiWifi" function does indeed fix the error and allows compile but it breaks the program as the WiFi hangs and does not connect.

khoih-prog, 2020-09-17:
This issue was mentioned and closed in

[**Compile Error on Platformio**](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/2)

It's **platform/compiler-settings dependent** and you can solve by adding the forward declaration as follows:

```
...
String separatorLine = "===============================================================";

uint8_t connectMultiWiFi(void);

//format bytes
String formatBytes(size_t bytes) 
...
```

Will update the examples soon to change.

closed 2020-09-17 by khoih-prog

khoih-prog, 2020-09-17:
Hi @tpalaz 

Just let you know the new [**ESPAsync_WiFiManager release v1.1.2**](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.1.2) already fixes the issue as well as some bugs in examples.