khoih-prog/ESPAsync_WiFiManager #12: Captive portal instead of static ip portal

geosys123, 2020-09-22:
Hi I am using following code for wifi manger picked from examples.
`ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "AutoConnectAP");
	ESPAsync_wifiManager.setDebugOutput(true);
	ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 100, 1), IPAddress(192, 168, 100, 1), IPAddress(255, 255, 255, 0));
	
	ESPAsync_wifiManager.setMinimumSignalQuality(-1);
	ESPAsync_wifiManager.setConfigPortalChannel(0);
	
	#if USING_CORS_FEATURE
		ESPAsync_wifiManager.setCORSHeader("Your Access-Control-Allow-Origin");
	#endif
	
	Router_SSID = ESPAsync_wifiManager.WiFi_SSID();
	Router_Pass = ESPAsync_wifiManager.WiFi_Pass();
	
	//Remove this line if you do not want to see WiFi password printed
	Serial.println("Stored: SSID = " + Router_SSID + ", Pass = " + Router_Pass);
	
	if (Router_SSID != "")
	{
		ESPAsync_wifiManager.setConfigPortalTimeout(120); //If no access point name has been previously entered disable timeout.
		Serial.println("Got stored Credentials. Timeout 120s");
	}
	else
	{
		Serial.println("No stored Credentials. No timeout");
	}
	
	String chipID = String(ESP_getChipId(), HEX);
	chipID.toUpperCase();
	
	// SSID and PW for Config Portal
	String AP_SSID = "ESP_" + chipID + "_AutoConnectAP";
	String AP_PASS = "12345678";
	
	if ( !ESPAsync_wifiManager.startConfigPortal(AP_SSID.c_str(), AP_PASS.c_str()) )
		Serial.println("Not connected to WiFi but continuing anyway.");
	else
		Serial.println("WiFi connected...yeey :)");`

The portal at **192.168.100.1** works fine when this code is called first time but if I call this code again in _loop_ or _setup_ function then it switches to captive portal which opens automatically when I connect to esp32 AP instead of portal at **192.168.100.1**. What is the reason for this behaviour?

khoih-prog, 2020-09-22:
@geosys123 

Have a look at these examples to know how to use Config Portal in loop()

1. [**Async_ConfigOnSwitch**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnSwitch)

2. [**Async_ConfigOnSwitchFS**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnSwitchFS)

3. [**Async_ConfigPortalParamsOnSwitch**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigPortalParamsOnSwitch)

4. [**Async_ConfigOnSwitchFS_MQTT_Ptr**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnSwitchFS_MQTT_Ptr)

I suggest you to **at least** go through all the examples to experience how and which is the best way to use the features of this [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) or any library.

The questions about issues which are already illustrated in examples or in previously Solved/Closed issues won't be answered next time as I don't have much time to deal with.

closed 2020-09-22 by khoih-prog

geosys123, 2020-09-22:
Sorry if I am not clear with my question. My question is I don't want captive portal but it is still opening captive portal when the code is called 2nd time in loop function. I want to use static IP portal which is working fine the code is called first time but on 2nd time it automaticaly changes to captive portal inspite of using `ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 100, 1), IPAddress(192, 168, 100, 1), IPAddress(255, 255, 255, 0));` in code.