khoih-prog/ESPAsync_WiFiManager #14: DHCP mode not working properly

prattom, 2020-09-24:
I am using Async_ConfigOnSwitch example but instead of static ip configuration I am using DHCP mode. Thus for this I am using following settings `#define USE_STATIC_IP_CONFIG_IN_CP          false` and ` #define USE_DHCP_IP     true` in modified code but the device always gets _255.255.255.255_ ip on connection to router. The router address or gateway address is _192.168.0.1_. The device is thus connected to router but I am still not able to access remote tcp server. This is happening with different router also.

khoih-prog, 2020-09-24:
@prattom 

Did you just use the example code ? 

If not, can you provide the minimized code to duplicate the problem? It's possible there is some wrong config somewhere.

Is the static IP working?

It's also better to try other examples as well to see if the problem is somewhere in your local network / devices.

prattom, 2020-09-24:
Yes I used exact same code  Async_ConfigOnSwitch except for settings `#define USE_STATIC_IP_CONFIG_IN_CP false` in the code. For static it works fine but not for DHCP mode. I tried other example also but same problem.

prattom, 2020-09-24:
Can't be a network problem since simple esp32 wifi code with just STA mode is working fine.

prattom, 2020-09-24:
Following is the log on first boot
>[WM] RFC925 Hostname = ConfigOnSwitch
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HELLO12307, Pass = 12345678
[WM] * Add SSID =  HELLO12307 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HELLO12307 , Router_Pass =  19801124
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: MTNL07 ,RSSI= -56
[WM] Channel: 4 ,IP address: 255.255.255.255
After waiting 4.12 secs more in setup(), connection result is connected. Local IP: 255.255.255.255
[WM] freeing allocated params!

Following is the log after pressing config switch and changing wifi settings
>Configuration portal requested.
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Opening configuration portal. Got stored Credentials. Timeout 120s
[WM] WiFi.waitForConnectResult Done
[WM] SET AP_STA
[WM] 
Configuring AP SSID = ESP_17286F24
[WM] AP PWD = 12345678
[WM] AP Channel = 9
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
E (140678) wifi: age_bss: null bss
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 0.70
[WM] Local ip = 255.255.255.255
[WM] Connection result:  WL_CONNECTED
connected...yeey :)
Local IP: 255.255.255.255
[WM] * Add SSID =  HDTTNL07 , PW =  12345678
[WM] * Add SSID =  HDTTNL07 , PW =  12345678
[WM] SaveWiFiCfgFile 
[WM] OK
[WM] freeing allocated params!

khoih-prog, 2020-09-24:
I retest the examples and they are all OK

### 1. Static IP with` #define USE_STATIC_IP_CONFIG_IN_CP false` commented out

```
Starting Async_ConfigOnSwitch using SPIFFS on ESP32_DEV
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] setSTAStaticIPConfig for USE_CONFIGURABLE_DNS
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -42
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 3.09 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
[WM] freeing allocated params!
```

### 2. DHCP with` #define USE_STATIC_IP_CONFIG_IN_CP false` enabled => OK

```
Starting Async_ConfigOnSwitch using SPIFFS on ESP32_DEV
[WM] RFC925 Hostname = ConfigOnSwitch
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -42
[WM] Channel: 2 ,IP address: 192.168.2.82
After waiting 4.46 secs more in setup(), connection result is connected. Local IP: 192.168.2.82
[WM] freeing allocated params!
HH
```

Did you notify the error in your terminal output

```
E (140678) wifi: age_bss: null bss
```

**It's possible something wrong / outdated in your system / hardware / firmware / ESP32 core**

See some similar issue of the ESP32 core in the past

1. [ WiFi and pppos co-exist problems (IDFGH-2548)](https://github.com/espressif/esp-idf/issues/4634)

Unless you can prove and I can duplicate there is some bug in the library, I'll close the issue now and you have to work by yourself to find out what the issue is.

closed 2020-09-24 by khoih-prog