khoih-prog/ESPAsync_WiFiManager #16: /close does not close the config portal

marcelstoer, 2020-09-24:
I'm using Async_ConfigOnSwitch  unchanged on ESP32.

When I click "Exit Portal" (i.e. request `/close`) the WiFi Manager remains in config mode until timeout is reached. This is unexpected to me. I expected the config portal to close and for the WiFi Manager to go back to station mode.

In order to exit the config portal I have to hit "Configuration" and then "Save" without entering any values.

Does this work as designed?

khoih-prog, 2020-09-24:
@marcelstoer 

Can you un-comment this [Line 1715](https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/src/ESPAsync_WiFiManager-Impl.h#L1715) from

```cpp
//stopConfigPortal = true; //signal ready to shutdown config portal		//KH crash if use this ???
```

to 

```cpp
stopConfigPortal = true; //signal ready to shutdown config portal		//KH crash if use this ???
```

to see if that is what you expect.

Possibly I forgot to un-comment after tested last time.

marcelstoer, 2020-09-25:
Indeed, that's the behavior I would expect. Thanks! In fact, on iOS the config portal window/dialog closes so quickly now that you don't even get to see the `/close` response "page". Not that I would care about it anyway 😄 Important is that the portal is shut down and that works nicely now.

khoih-prog, 2020-09-25:
That's good. Will update in next release.

closed 2020-09-25 by khoih-prog

marcelstoer, 2020-09-27:
Can I ask you to keep issues open until the respective fix has landed on `master`? That would allow us to be notified by GitHub when we can update our dependencies.

khoih-prog, 2020-09-28:
Master updated.

khoih-prog, 2020-10-16:
See [ESP_WiFiManager v1.2.0](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.2.0) and [Contributions-and-Thanks](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

### Releases v1.2.0

1. Restore cpp code besides Impl.h code to use in case of `multiple definition` linker error. See [`Change Implementation to seperate *.h and *.cpp file instead of *.h and *-Impl.h`](https://github.com/khoih-prog/ESP_WiFiManager/issues/38) and [`Support building in PlatformIO PR`](https://github.com/khoih-prog/ESP_WiFiManager/pull/20). Also have a look at [**HOWTO Fix Multiple Definitions Linker Error**](https://github.com/khoih-prog/ESP_WiFiManager#HOWTO-Fix-Multiple-Definitions-Linker-Error)
2. Fix bug [/close does not close the config portal](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/16).