khoih-prog/ESPAsync_WiFiManager #19: ESP32 static IP not saved after restarting the device

krupis, 2020-11-12:
Hello. I am learning about esp32 and trying your library. I am using the example program:
Async_ESP32_FSWebServer_DRD

I have not modified the code yet. When I run the program. A new network appeared 
[image]

I am able to connect to my device by typing the ip address :
192.168.1.100
and change some configurataion:

Now I can see that the device connects with the new parameters: (I have set a new static ip 192.168.4.210)

[image]

However, the thing that I do not fully understand, when i restart my esp32 device, it goes changes my static ip (192.168.4.210) to:
192.168.2.232
[image]

Is that normal? Should it not save my configuration even after I restart the device?

Also, I am unsure about: 
Open http://async-esp32fs.local/edit to see the file browser
 Using username = admin and password = admin

Opening this webpage in a browswer fails everytime. How do i use it?
[image]

Appreciate any sort of advice. Thanks in advance.

khoih-prog, 2020-11-12:
Hi @krupis 

Thanks for using the library.

> Is that normal? Should it not save my configuration even after I restart the device?

Currently, the example still is not perfect, with the bug as you experienced (Static IP input in CP is not persistent).
You can **hard-code staticIP** in the sketch as follows:

```
#ifdef ESP32
    IPAddress stationIP   = IPAddress(192, 168, 4, 210);     // Change this to persistent Static IP
  #else
    IPAddress stationIP   = IPAddress(192, 168, 4, 186);
  #endif
```

> Opening this webpage in a browswer fails everytime. How do i use it?

If the ESP32 MDNS is somehow not working perfectly (depending on your network, OS, etc.), and you can use **IP address directly** to access the Server, for example: `http://192.168.4.210/edit` instead of `http://async-esp32fs.local/edit `.

krupis, 2020-11-12:
Thanks for your quick response. I really need to be able to change the ip
through the server because once device is programmed it will not be touched
again so perhaps il have to dig deeper see if I can troubleshoot. Also, is
it possible to change from the static IP mode to dhcp mode?

khoih-prog, 2020-11-12:
I really advise you against the use of staticIP unless you have good reason to do so. StaticIP will create lots of problem if being used not correctly, especially for normal users, and users certainly have to know more about the network to set it correctly, without errors and conflicts.

It's better for you to use DHCP now, by modifying the code as follows:

```
// Use USE_DHCP_IP == true for dynamic DHCP IP, false to use static IP which you have to change accordingly to your network
#if (defined(USE_STATIC_IP_CONFIG_IN_CP) && !USE_STATIC_IP_CONFIG_IN_CP)
// Force DHCP to be true
  #if defined(USE_DHCP_IP)
    #undef USE_DHCP_IP
  #endif
  #define USE_DHCP_IP     true
#else
  // You can select DHCP or Static IP here
  #define USE_DHCP_IP     true                      // Always using DHCP  <=====================
  //#define USE_DHCP_IP     false
#endif
```

The CP is now as simple as this and much better for users

[image]

and you can see MDNS will work correctly with DHCP

[image]

[image]

krupis, 2020-11-12:
That is true but the project I am doing must have an optiom to do so. I may
not need it most of the time but I still need to be able to switch to
static mode if required. For example each esp will be given an IP on a
seperate network just for monitoring temperatures. I may need to know every
device IP address and be sure that it will not change since I will ise
modbus tcp to transfee data

khoih-prog, 2020-11-12:
If you use DHCP, you can easily assign every device a **unique MDNS name** so that MODBUS TCP can reach to xfer data. 

If staticIP is selected, that fastest way to do now for you, before the feature is fixed, is to use dynamic parameter (**ESPAsync_WMParameter**) to input from CP/ store staticIP, then replacing staticIP, from [Line 486](https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/examples/Async_ESP32_FSWebServer_DRD/Async_ESP32_FSWebServer_DRD.ino#L486) with the dynamic param (staticIP) got from CP.

```
#if !USE_DHCP_IP
  #if USE_CONFIGURABLE_DNS
    // Set static IP, Gateway, Subnetmask, DNS1 and DNS2.
    ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask, dns1IP, dns2IP);
  #else
    // Set static IP, Gateway, Subnetmask, Use auto DNS1 and DNS2.
    ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask);
  #endif
#endif
```

Have a look at other examples about how to use dynamic parameters **ESPAsync_WMParameter**.

You have to understand that these are just examples to demonstrate the basic operation of the library, and there is no one-size-fit-all solution for every use case.

If you can find a quick solution, please post here to help other library's users.

khoih-prog, 2020-12-21:
This has been fixed in [**ESPAsync_WiFiManager v1.4.0]**(https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.4.0)
---

### Major Releases v1.4.0

1. Fix staticIP not saved in examples. See [ESP32 static IP not saved after restarting the device](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/19)
2. Add structures and functions to handle AP and STA IPs.
3. Add complex examples 
  * [**Async_ConfigOnDRD_FS_MQTT_Ptr_Complex**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_FS_MQTT_Ptr_Complex) to demo usage of `std::map`
  * [**Async_ConfigOnDRD_FS_MQTT_Ptr_Medium**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_FS_MQTT_Ptr_Medium).
4. Add simple minimal examples 
  * [**Async_ConfigOnDRD_ESP32_minimal**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_ESP32_minimal)
  * [**Async_ConfigOnDRD_ESP8266_minimal**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_ESP8266_minimal)
  * [**Async_AutoConnect_ESP32_minimal**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect_ESP32_minimal)
  * [**Async_AutoConnect_ESP8266_minimal**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect_ESP8266_minimal)
5. Modify Version String
6. Add Table of Contents

---

Your contribution has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

closed 2020-12-21 by khoih-prog