khoih-prog/ESPAsync_WiFiManager #20: How is the SSID and Password being written to SPIFFS?

thewhiterabbit, 2020-11-22:
Thank you for this great tool.

I need to add more stored data to be entered on the configuration page.

Where do I add a new code block for another data entry?

Where do I edit the HTML to customize the configuration page?

khoih-prog, 2020-11-22:
Thanks for using the library.

I suggest you **at least** go through all the examples to have the experience of **what and how to use the features** this [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) is providing.

Then you will know that to add new configurable data, you don't need to change the library source code or modify any HTML. But just easily use **ESPAsync_WMParameter**, as in [**Async_AutoConnectWithFSParameters** example](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnectWithFSParameters) to add 5 new configurable parameters.

```
bool loadFileFSConfigFile(void)
{
   ...
}

bool saveFileFSConfigFile(void)
{
   ...
}

void setup()
{
...
// The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  ESPAsync_WMParameter custom_blynk_server("blynk_server", "blynk_server", blynk_server, BLYNK_SERVER_LEN + 1);
  ESPAsync_WMParameter custom_blynk_port  ("blynk_port",   "blynk_port",   blynk_port,   BLYNK_SERVER_PORT_LEN + 1);
  ESPAsync_WMParameter custom_blynk_token ("blynk_token",  "blynk_token",  blynk_token,  BLYNK_TOKEN_LEN + 1 );

  ESPAsync_WMParameter custom_mqtt_server("mqtt_server", "mqtt_server", mqtt_server, MQTT_SERVER_MAX_LEN + 1);
  ESPAsync_WMParameter custom_mqtt_port  ("mqtt_port",   "mqtt_port",   mqtt_port,   MQTT_SERVER_PORT_LEN + 1);
  ...
}
```

Certainly, you have to learn and do that yourself, as I can't have time to help you do your research and code.

closed 2020-11-22 by khoih-prog

thewhiterabbit, 2020-11-22:
Negative.

I don't want to connect to Blynk.

I want to store data to SPIFFS/EEPROM.

That's why I asked where to modify the code to customize the HTML and add data storage block for my own settings, not Blynk server login and address data...

khoih-prog, 2020-11-22:
Example is just an example, and you must modify according to your use-case.
You certainly have to modify `Blynk-related` data to what you need. And at least basic knowledge is necessary to use the library.

No more answer until you read and understand what I meant.

Please don't write nonsense here unless you understand what you are doing.

thewhiterabbit, 2020-11-22:
Identify the file and line number of the EEPROM/SPIFFS code blocks that store data in the register (if you didn't steal it from another library and credit yourself). I found the HTML.

thewhiterabbit, 2020-11-23:
> Thanks for using the library.
>
> I suggest you **at least** go through all the examples to have the experience of **what and how to use the features** this [...]

^This only tells me that you can add parameters, not how to add them. I assume that you use .addparameter or .newparameter, however, there is a lot of fog in your "explanation". Why is there max length constants? Do I even need the max length constants? Where is the parameter add supposed to take place? Do I need to be inside the configfilewrite function to add parameters?

Forgive me for asking questions. You must be vastly superior to me in every way and have never ever needed anyone's help, ever. Well, I'm not you. I need a little help adding these parameters because your code lacks documentation and sufficient explanatory commenting. Had you written a documentation I would not be bothering you whatsoever. You know about RTFM? Well, WTFM! (w=write)... I thought people in the open source maker community were all about helping others learn. I thought wrong, apparently.

khoih-prog, 2020-11-23:
I'm so sorry if you're always thinking the wrong way.

I just hope we're not **ungrateful teenagers** anymore, always blaming other people problems happening because of our faults.

Many of us have been spending hard time to learn and getting our knowledge, then experiencing by investing to get knowledge once, we'll harvest the good fruit **your whole life**. That's the ideal target we'll like to have, otherwise, why wasting our time here. So asking you to spend some time to research, to go thru examples, etc. , then from that to have experience, before blaming, is not excessive.

The open source community will help you, by creating many facilities for you to learn, use and save time, not by giving somebody a way to cheat (I mean doing things for you, or giving you some kind of fast food that will harm you in the future, etc.). Especially we're not a **code factory.**

I've tried my best to WT most complete possible FM, but certainly up to certain level, and many people so far are OK and successfully using the library. 

If you think you can rewrite a better manual and contribute back to the community, you're very welcome. Just don't sit and blame everybody around you.

I'll close and delete this thread as it's not helpful and negative.