khoih-prog/ESPAsync_WiFiManager #22: When using custom parameters root config page is available but all other config pages are blank.

thewhiterabbit, 2020-11-23:
I am willing to contribute. I just need to know where to add the custom parameters. In your documentation you don't say where to add them. I tried to add them in the same place as your example (loop, after the instantiation of ESPAsync_wifiManager), however the web page /wifi becomes blank after adding the custom parameter via copy paste exactly as it is in the example file.

So, if you will elaborate I will amend the documentation and pull request. I will also amend any other parts that were difficult for me to figure out and make it clear in the documentation where to add and why.

`    //
    // START WEB SERVER LOOP
    //
    // is configuration portal requested?
    if ((digitalRead(TRIGGER_PIN) == LOW))
    {
        Serial.println("\nConfiguration portal requested.");
        digitalWrite(LED_BUILTIN, LED_ON); // turn the LED on by making the voltage LOW to tell us we are in configuration mode.

        //Local intialization. Once its business is done, there is no need to keep it around
        ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "AsyncCP-ParamsOnSW");

        char customhtml[24] = "type=\"checkbox\""; //<--This is where I copied from
        bool sensorDht22 = true;
        if (sensorDht22)
        {
            strcat(customhtml, " checked");
        }
        ESPAsync_WMParameter p_sensorDht22("DHT-22", "DHT-22 Sensor", "T", 2, customhtml, WFM_LABEL_AFTER);
        ESPAsync_wifiManager.addParameter(&p_sensorDht22); //<-- This is the end of the copy/paste

        //Check if there is stored WiFi router/password credentials.
        //If not found, device will remain in configuration mode until switched off via webserver.
        Serial.print("Opening configuration portal. ");
        
        Router_SSID = ESPAsync_wifiManager.WiFi_SSID();
        Router_Pass = ESPAsync_wifiManager.WiFi_Pass();
        
        // From v1.1.0, Don't permit NULL password
        if ( (Router_SSID != "") && (Router_Pass != "") )
        {
        ESPAsync_wifiManager.setConfigPortalTimeout(120); //If no access point name has been previously entered disable timeout.
        Serial.println("Got stored Credentials. Timeout 120s");
        }
        else
        Serial.println("No stored Credentials. No timeout");

        // Extra parameters to be configured
        // After connecting, parameter.getValue() will get you the configured value
        // Format: <ID> <Placeholder text> <default value> <length> <custom HTML> <label placement>

        // Config Portal SSID - this is a straight forward string parameter
        ESPAsync_WMParameter p_PortalSSID(PortalSSID_Label, "Portal_SSID", PortalSSID, SSID_MAX_LENGTH + 1);

        // Config Portal SSID - this is a straight forward string parameter
        ESPAsync_WMParameter p_PortalPassword(PortalPassword_Label, "Portal_Password", PortalPassword, PASSWORD_MAX_LENGTH + 1);

        //add all parameters here
        ESPAsync_wifiManager.addParameter(&p_PortalSSID);
        ESPAsync_wifiManager.addParameter(&p_PortalPassword);

        // Sets timeout in seconds until configuration portal gets turned off.
        // If not specified device will remain in configuration mode until
        // switched off via webserver or device is restarted.
        //ESPAsync_wifiManager.setConfigPortalTimeout(120);

        ESPAsync_wifiManager.setMinimumSignalQuality(-1);

        // From v1.0.10 only
        // Set config portal channel, default = 1. Use 0 => random channel from 1-13
        ESPAsync_wifiManager.setConfigPortalChannel(0);
        //////

        //set custom ip for portal
        //ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 100, 1), IPAddress(192, 168, 100, 1), IPAddress(255, 255, 255, 0));
        
    #if !USE_DHCP_IP    
    #if USE_CONFIGURABLE_DNS  
        // Set static IP, Gateway, Subnetmask, DNS1 and DNS2. New in v1.0.5
        ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask, dns1IP, dns2IP);  
    #else
        // Set static IP, Gateway, Subnetmask, Use auto DNS1 and DNS2.
        ESPAsync_wifiManager.setSTAStaticIPConfig(stationIP, gatewayIP, netMask);
    #endif 
    #endif  

    // New from v1.1.0
    #if USING_CORS_FEATURE
    ESPAsync_wifiManager.setCORSHeader("Your Access-Control-Allow-Origin");
    #endif

        // Start an access point and goes into a blocking loop awaiting configuration.
        // Once the user leaves the portal with the exit button
        // processing will continue

        static bool resultConfigPortal;
        if ( (PortalSSID[0] == 0) || (PortalPassword[0] == 0) )
        {
        resultConfigPortal = ESPAsync_wifiManager.startConfigPortal((const char *) DefaultPortalSSID.c_str(), DefaultPortalPassword.c_str());
        }
        else
        {
        resultConfigPortal = ESPAsync_wifiManager.startConfigPortal((const char *) PortalSSID, PortalPassword);
        }

        if (resultConfigPortal)
        {
        Serial.println("WiFi connected...yeey :)");
        Serial.print("Local IP: ");
        Serial.println(WiFi.localIP());
        }
        else
        Serial.println("Not connected to WiFi but continuing anyway.");

        // Only clear then save data if CP entered and with new valid Credentials
        // No CP => stored getSSID() = ""
        if ( String(ESPAsync_wifiManager.getSSID(0)) != "" && String(ESPAsync_wifiManager.getSSID(1)) != "" )
        {
        // Stored  for later usage, from v1.1.0, but clear first
        memset(&WM_config, 0, sizeof(WM_config));
        
        for (uint8_t i = 0; i < NUM_WIFI_CREDENTIALS; i++)
        {
            String tempSSID = ESPAsync_wifiManager.getSSID(i);
            String tempPW   = ESPAsync_wifiManager.getPW(i);
        
            if (strlen(tempSSID.c_str()) < sizeof(WM_config.WiFi_Creds[i].wifi_ssid) - 1)
            strcpy(WM_config.WiFi_Creds[i].wifi_ssid, tempSSID.c_str());
            else
            strncpy(WM_config.WiFi_Creds[i].wifi_ssid, tempSSID.c_str(), sizeof(WM_config.WiFi_Creds[i].wifi_ssid) - 1);
        
            if (strlen(tempPW.c_str()) < sizeof(WM_config.WiFi_Creds[i].wifi_pw) - 1)
            strcpy(WM_config.WiFi_Creds[i].wifi_pw, tempPW.c_str());
            else
            strncpy(WM_config.WiFi_Creds[i].wifi_pw, tempPW.c_str(), sizeof(WM_config.WiFi_Creds[i].wifi_pw) - 1);  
        
            // Don't permit NULL SSID and password len < MIN_AP_PASSWORD_SIZE (8)
            if ( (String(WM_config.WiFi_Creds[i].wifi_ssid) != "") && (strlen(WM_config.WiFi_Creds[i].wifi_pw) >= MIN_AP_PASSWORD_SIZE) )
            {
            LOGERROR3(F("* Add SSID = "), WM_config.WiFi_Creds[i].wifi_ssid, F(", PW = "), WM_config.WiFi_Creds[i].wifi_pw );
            wifiMulti.addAP(WM_config.WiFi_Creds[i].wifi_ssid, WM_config.WiFi_Creds[i].wifi_pw);
            }
        }
        
        saveConfigData();
        }

        // Getting posted form values and overriding local variables parameters
        // Config file is written regardless the connection state
        strcpy(PortalSSID, p_PortalSSID.getValue());
        strcpy(PortalPassword, p_PortalPassword.getValue());

        // Writing JSON config file to flash for next boot
        writeConfigFile();

        digitalWrite(LED_BUILTIN, LED_OFF); // Turn LED off as we are not in configuration mode.
    }
    check_status();
    //
    // END WEB SERVER LOOP
    //`

When using the above code the root config page is available but all other pages are blank.

Where do the custom parameters need to be located in the loop to function properly?

khoih-prog, 2020-11-23:
This sounds much better and I'll have a look at that issue very soon when I have time as I'm currently working on something now. To be done within several days.

In the mean time, just explore some more the examples and post as much info as possible to save time for us to duplicate.
You'll benefit a good deal from that well-spent time.

thewhiterabbit, 2020-11-23:
Okay, thank you.

I also tried to use radio elements like this...
```
ESPAsync_WMParameter customField_C("<input name="tempMode" type="radio" id="celsius" value=1><label for="celsius">Celsius</label>");
ESPAsync_wifiManager.addParameter(&customField_C);

ESPAsync_WMParameter customField_F("<input name="tempMode" type="radio" id="fahrenheit" value=2><label for="fahrenheit">Fahrenheit</label>");
ESPAsync_wifiManager.addParameter(&customField_F);

```

but this approach must be completely incorrect because the data in value=[1 or 2] is never stored in the FS key "tempMode".

I would like to add this radio button example to you documentation for pull once I get it working.

If you need help on your other project I can help.

khoih-prog, 2020-11-23:
I'm in final steps to test after finishing this to help a person on Arduino Forum. Only one more example to do.

[**TimerInterrupt for Arduino Nano 33 BLE**](https://forum.arduino.cc/index.php?topic=714757.0)

Do you have Nano-33-BLE to test?

If you're willing to participate, I'll coordinate with you to test new Libraries and Libraries' releases, to save me some time as I still have to do it mostly alone, sometimes with the help of [**Miguel Wisintaine**r](https://github.com/tcpipchip), a Univ. Prof. in Brazil.

I believe you'll get a lot more experience thru that, and know a lot deeper details of many libraries. That'll very good to be used in the future.

thewhiterabbit, 2020-11-23:
No, I don't have that board. :(

I can order different boards for testing though. Yes, I'd like to help in any way I can.

I'm working on a way to iterate through all custom parameters added and save to file instead of having to add them in both `writeConfig` and `readConfig` functions like in your examples. Maybe core can be modified to array everything added with `.addParameter` and automatically save with a method like `.saveParametersFS()`.

I'm very good at engineering improvements once I know how cores work.

khoih-prog, 2020-11-23:
The current way to add params is very clumsy, and I'll modify this [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) and [**ESP_WiFiManager**](https://github.com/khoih-prog/ESP_WiFiManager) very soon to use the solution similar to these libraries

Look at how I implemented dynamic params:

1. [**Blynk_WM**](https://github.com/khoih-prog/Blynk_WM)
2. [**BlynkEthernet_WM**](https://github.com/khoih-prog/BlynkEthernet_WM)
3. [**WiFiManager_NINA_Lite**](https://github.com/khoih-prog/WiFiManager_NINA_Lite)

Happy to see that you have engineering background. Me too but I'm retiree now.

and many more.

khoih-prog, 2020-11-23:
I'm done with and already posted new  [**NRF52_MBED_TimerInterrupt** Library](https://github.com/khoih-prog/NRF52_MBED_TimerInterrupt) and on Arduino Forum as [**NRF52_MBED_TimerInterrupt** Library for Nano-33-BLE](https://forum.arduino.cc/index.php?topic=715019.0)

You're very welcome to test it whenever you have the **Nano-33-BLE.**

Will have a look at your issue late tomorrow.

thewhiterabbit, 2020-11-23:
Okay.

I like the way dynamic parameters work in those linked repos.

I just encountered a problem with adding option type inputs currently; They need the same name, but different IDs for label elements.

My solution is to implement `.addParameter` with the` element ID` value as an option, and it will default to the same value as `element Name` if unspecified. this way option elements work and keep the same name, with different IDs.

I'll pull request tomorrow when I get done amending `ESPAsync_WiFiManager-Impl.h`.

khoih-prog, 2020-11-23:
I don't think we'll make **disruptive** change to this library as many people are using and relying on it.

I'll create a new one to use the **new and easy dynamic params** configuration as you've seen. Already done that new lib halfway quite some time ago, then left it there to rust.

So **don't make disruptive PR** now to this lib, and wait for the new one to come. 

Just concentrate on the documentation and bug for **this lib and** its twin [**ESP_WiFiManager**](https://github.com/khoih-prog/ESP_WiFiManager) now

khoih-prog, 2020-11-24:
Just invite you into Private Repo for later discussion about future projects, collaboration and tests.

Have a look at the first unfinished draft of  [**HOWTOAddDynamicParameters**](https://github.com/khoih-prog/ESPAsync_WiFiManager_Private/blob/main/HOWTOAddDynamicParameters.md) and modify as necessary.

Fresh outsiders always have better vision that insiders because of tunnel-vision.

khoih-prog, 2020-11-24:
The code you post at the beginning is running OK on all pages.

[image]

[image]

closed 2020-11-24 by khoih-prog