khoih-prog/ESPAsync_WiFiManager #23: Non-blocking mode

LuckHermsen, 2020-11-28:
Any chance the non-blocking option will be available soon?
I was just added to the development branch of the original WiFiManager: https://github.com/tzapu/WiFiManager/tree/development

I think this would be very useful for everyone running devices that should be able to fully operate in STA and AP mode but give the user the ability to switch between those 2 modes on demand.

My programming skills are limited, but if I can help/support to implement this feature I'd be happy to help.

khoih-prog, 2020-11-28:
> Any chance the non-blocking option will be available soon?

Can you clarify what is this and give some example.
The WiFiManager is just a tool to permit you to input certain Credentials in start-up. We don't like to have it running all the times to create conflicts and to interfere with other jobs and use precious resource, unless absolutely necessary.

Check these closed issues

1. [**Wifi Manager acts like Home Wifi Router?**](https://github.com/khoih-prog/ESP_WiFiManager/issues/12)
2. [**Accessing manager after connection**](https://github.com/khoih-prog/ESP_WiFiManager/issues/15)

> I dropped the persistent Config Portal quite a while and use dynamic Config Portal with these features:
> 
> open only when necessary to enter Credentials
> closed whenever Credentials are valid and Config Portal not needed
> More efficient as Config Portal WebServer is not running in normal operation.
> because of these following concerns and potential issues
> 
> AP stays open
> Wifi Manager acts like Home Wifi Router?
> Imagine you have 20 different devices, each running its own AP. There will be a lot of issues such as :
> 
> channel conflict => big issue for every board, even for your router if channel is duplicated.
> unintended (re-)connection from other devices
> the STA+AP mode is not as very reliable as STA only.
> the Portal Webserver must be still running and not efficient to do so.

---

> I think this would be very useful for everyone running devices that should be able to fully operate in STA and AP mode but give the user the ability to switch between those 2 modes on demand

Can you give some use case why we need this option? When the board has no WiFi Credential, the Config Portal AP mode is a must and is necessary, only for a very short time, for you to input Credentials. The STA Config Portal is only possible after the board has correct Credentials and has connected to the WiFi. Leaving the CP running is not advisable.

---

> My programming skills are limited, but if I can help/support to implement this feature I'd be happy to help.

Your contribution is always welcome anytime. The more important assets here are "your novel thinkings and problem solutions", not "programming skills", and don't hesitate to propose new solutions, etc.

If you're willing to contribute, I'll invite you to a **private repo** where we can participate and contribute ideas, solutions, etc to all libraries.

Regards,

LuckHermsen, 2020-11-30:
Hi Khoih,

What I would like to do is integrate the Async WiFiManager into my existing project, but without blocking my main code.
The program I've written is meant to be used in cars. Some people will have a Hotspot in their car, others might not. In the latter case I want the ESP to be in AP mode and run all other code, just like it would in STA mode. 
Should the user want to switch between AP or STA mode, it would be perfect if the user can navigates to a specific page (i.e. /settings) were the /wifi page will be displayed, allowing the user to pick a WiFi network.

I tried commenting all server handles except '/wifi' in the WiFimanager library, but my 'own' server.on events wouldn't work.
Maybe I could create the WiFiManager instance only when the user accesses the settings page.. But it would still be nice if the main code doesn't block.

Any ideas on how best to approach this?

khoih-prog, 2020-11-30:
> In the latter case I want the ESP to be in AP mode and run all other code, just like it would in STA mode.
> Should the user want to switch between AP or STA mode, it would be perfect if the user can [...]

In programming world, anything is possible provided that you know how to solve the issue. To achieve this, I suggest you have to spend more time to look at the library, the example code and test then retest until **you master the usage and configuration.** You also have to know the **purpose** of the library, its **limitation** and where/how to use it effectively.

As I said,

> The WiFiManager is just a tool to permit you to input certain Credentials in start-up.

**The library has nothing to do and should not interfere with your application code, until you push it into operation, by certain conditions (double reset, SW pressed, can't connect to programmed AP, etc).**

If you don't like to have the **blocking** `ESPAsync_WiFiManager` running is start-up. By design, **without Credentials / WiFi AP, `ESPAsync_WiFiManager`  has to block and wait for you to enter correct Credentials to connect to a WiFi AP.**

In your use case, without AP, you can just ignore `ESPAsync_WiFiManager` by not calling it at all in setup(). Then you run your normal code as usual, certainly using **(Async)WebServer if necessary**, **but in AP mode.** Users have to connect to that AP, get an IP address from its DHCP server, to access the (Async)WebServer. 

Nothing to do with `ESPAsync_WiFiManager` so far because it's not activated yet or has been killed.

Whenever you need to change the mode, e.g. to STA from AP if the car has local HotSpot, just start the `ESPAsync_WiFiManager` by a hardware SW or just a virtual `BUTTON` control by your WebServer.

Have a look at [**Async_ConfigOnSwitch**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnSwitch) loop() to know how to start the `ESPAsync_WiFiManager` in that case.

I'm afraid I can't spend more time to help solve your problem anymore. If I have time in the near future, I possibly write an example to do something for similar use-case.

Good luck with your project.

closed 2020-11-30 by khoih-prog