khoih-prog/ESPAsync_WiFiManager #24: typedef declaration conflicts

malves61, 2020-12-15:
Khoih,

Please be patient, I'm new on platformio and github, so this is probably a very dumb question and not a real issue. 

I started experimenting with ESPAsync_WiFiManager. I am having conflicting typedef declaration when trying to build the example Async_ConfigOnSwitch.ino (renamed it main.cpp on vscode). The declarations are on the underlying libraries, please see the build output below.

Do you see anything that I might have missed? Any hints?

Thank you

malves61
```

> Executing task in folder ESPbrute-digital-sensor-relay-MQTT: pio run <

Processing ESP8266 (platform: espressif8266; framework: arduino; board: d1_mini_pro)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.7 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP_DoubleResetDetector> 1.0.3
|   |-- <LittleFS(esp8266)> 0.1.0
|-- <OneButton> 0.0.0-alpha+sha.eb583d713a
|-- <ArduinoJson> 6.17.2
|-- <Adafruit MQTT Library> 2.1.0
|-- <ESPAsync_WiFiManager> 1.3.0
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.17.2
|   |-- <ESP_DoubleResetDetector> 1.0.3
|   |   |-- <LittleFS(esp8266)> 0.1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <WiFi101> 0.16.1
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <WiFi101> 0.16.1
|-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <WiFi101> 0.16.1
Building in release mode
Compiling .pio/build/ESP8266/src/main.cpp.o
src/main.cpp:324:4: warning: #warning Using DHCP IP [-Wcpp]
   #warning Using DHCP IP
    ^
Compiling .pio/build/ESP8266/libb94/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
In file included from .pio/libdeps/ESP8266/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.h:597:0,
                 from src/main.cpp:347:
.pio/libdeps/ESP8266/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager-Impl.h:994:3: warning: multi-line comment [-Wcomment]
[... 75 lines cut ...]
Indexing .pio/build/ESP8266/libb94/libESP8266WiFi.a
Compiling .pio/build/ESP8266/lib8af/ESP Async WebServer/SPIFFSEditor.cpp.o
Compiling .pio/build/ESP8266/lib8af/ESP Async WebServer/WebAuthentication.cpp.o
Compiling .pio/build/ESP8266/lib8af/ESP Async WebServer/WebHandlers.cpp.o
Archiving .pio/build/ESP8266/libe61/libESPAsyncTCP.a
Indexing .pio/build/ESP8266/libe61/libESPAsyncTCP.a
Compiling .pio/build/ESP8266/lib8af/ESP Async WebServer/WebRequest.cpp.o
Archiving .pio/build/ESP8266/lib607/libHash.a
Indexing .pio/build/ESP8266/lib607/libHash.a
Compiling .pio/build/ESP8266/lib8af/ESP Async WebServer/WebResponses.cpp.o
*** [.pio/build/ESP8266/src/main.cpp.o] Error 1
In file included from .pio/libdeps/ESP8266/ESP Async WebServer/src/SPIFFSEditor.cpp:1:0:
.pio/libdeps/ESP8266/ESP Async WebServer/src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated (declared at /home/marcelo/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
                                                                                                     ^
====================================================================== [FAILED] Took 3.89 seconds ======================================================================

Environment    Status    Duration
-------------  --------  ------------
ESP8266        FAILED    00:00:03.886
================================================================ 1 failed, 0 succeeded in 00:00:03.886 ================================================================
The terminal process "pio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

khoih-prog, 2020-12-15:
There is something wrong with your library usage.

The ESP8266 doesn't need `WiFi101` library. Adding it will create sure conflict errors.

Just remove all instances of `WiFi101` and it'll be OK.

```
|-- <DNSServer> 1.1.1
|   |   |-- <WiFi101> 0.16.1
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <WiFi101> 0.16.1
|-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <WiFi101> 0.16.1
```

It's also possible you don't use the correct DNSServer library for ESP8266. Be sure to check.

closed 2020-12-15 by khoih-prog

khoih-prog, 2020-12-15:
Another note is that the `DNSServer` library is included in the ESP8266/ESP32 cores

1. `ESP8266 built-in library` => [DNSServer Library](https://github.com/esp8266/Arduino/tree/master/libraries/DNSServer)
2. `ESP32 built-in library` . => [DNSServer Library](https://github.com/espressif/arduino-esp32/tree/master/libraries/DNSServer)

You don't need to download any new  `DNSServer` library at all.

malves61, 2020-12-15:
Thanks for your patience and kind answer. 
I found WiFi101 referenced at c_cpp_properties.json on the .vscode folder - probably I made a mistake with Intellisense. 
Deleting .vscode and all libraries from the project and adding   khoih-prog/ESPAsync_WiFiManager @ ^1.3.0 to the dependencies did the trick.

khoih-prog, 2020-12-15:
That's great. Your complete bug report is very helpful to locate the issue and remember doing the same way next time (hopeful no more issue;=))

Without complete picture, no one can have psychic power to guess what's wrong, so it'll be time-consuming to solve, even the smallest issue.

Cheers,