khoih-prog/ESPAsync_WiFiManager #25: Documentation lacking minimal example

positron96, 2020-12-20:
Hello and thanks for the library. 

The original ESP8266 WifiManager library can be used with default settings using 3 lines of code: 1 include and 2 lines in setup(). (See example: https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnect). The example with same name in ESPAsync_WiFiManager has whooping 500 lines of code, and it's not immediately clear what it does and what to pull in my code to have default functionality, e.g. start captive portal and connect the board to WiFi without having to hardcode the ESSID/password. 
Please, provide a minimal example of this typical usecase, if one is possible.

khoih-prog, 2020-12-21:
I just hope anybody using this library spending some time to look at the examples, understand and test them to have much better knowledge that will definitely benefit in the long run.

I really don't expect any user uses as `fast food`, without spending any time with it. The versatility and functionality are what I like to provide to the library's users, not just simplicity without any long term benefit.

Anyway, I've just  created several dead-simple examples which I just hope anybody could do, simply by deleting the unwanted portions of code

---

#### Simplest Possible

 1. [Async_AutoConnect_ESP32_minimal](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect_ESP32_minimal)
 2. [Async_AutoConnect_ESP8266_minimal](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect_ESP8266_minimal)
 3. [Async_ConfigOnDRD_ESP32_minimal](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_ESP32_minimal) 
 4. [Async_ConfigOnDRD_ESP8266_minimal](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDRD_ESP8266_minimal)

closed 2020-12-21 by khoih-prog

positron96, 2020-12-21:
Amazing, thank you!