khoih-prog/ESPAsync_WiFiManager #27: Not working when not connected to Serial

SkyGuardian42, 2020-12-21:
The WiFi Manager doesn't work when powering the ESP32 over its VIN Port, but does when powering it through USB. I'm simply using the Code from the Async_AutoConnect_ESP32_minimal.ino example. I already remove the `while (!Serial); delay(200);` line. Is there a simple fix for this?

SkyGuardian42, 2020-12-21:
Turns out there is a more general problem with powering the ESP32 on the VIN Pins, which has nothing to do with the library.

closed 2020-12-21 by SkyGuardian42

khoih-prog, 2020-12-21:
I've just been back home and seen this and am happy **you did your thorough research/experiment** and found out the issue before I would inform you.

This is good experience for other new users, and will be definitely helpful. Don't hesitate to post any issue, **correct or not**, as they are all beneficial somehow to other users.

Good Luck with your project,