khoih-prog/ESPAsync_WiFiManager #28: Error esp_littlefs.c  'utime_p'

solroshan, 2020-12-23:
Hello Khoih,
First of all thank you for this great library shared with us.
If i can, i'll gonna use your library for my home project but i could not build any examples. Checked closed issues for fix by myself but im newbie little bit, maybe you know the way to fix.
Im using PlatformIO on Atom in ubuntu.

This is my platformio.ini;
[image]

Im not writing all this 665 lines for not to pollute the post //And dont know how can I write beautiful yet.
Example is Async_AutoConnect.ino under the "examples" i just copied and pasted with any changes.
Build is giving this results;
[image]
I can see something wrong with all littlefs but don't know how to fix.
I want to autoconnect my iot and watching online sensor data am I on right way? :))

solroshan, 2020-12-23:
Hello again, I'm here to close this issue.

Solution shared by littlefs library owner, @lorol.

go to LittleFS_esp32 /src/esp_littlefs.c
and uncomment line **#define CONFIG_LITTLEFS_FOR_IDF_3_2**

closed 2020-12-23 by solroshan

khoih-prog, 2020-12-23:
Thanks for your using this library and your nice words.

I think there is something wrong with your **PIO installation** and/or the PIO version of very-new [**LittleFS_esp32**](https://github.com/lorol/LITTLEFS). It's really difficult to guess which.

I suggest you try these options:

1. Using the AutoConnect minimal version [**Async_AutoConnect_ESP32_minimal**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect_ESP32_minimal)
2. Using default ESP32's SPIFFS by modifying the code [**Async_AutoConnect.ino#L54-L55**](https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/examples/Async_AutoConnect/Async_AutoConnect.ino#L54-L55) 

from

```
#define USE_LITTLEFS    true
#define USE_SPIFFS      false
```

to

```
#define USE_LITTLEFS    false
#define USE_SPIFFS      true
```

It's also advisable that you use the Arduino IDE

Good Luck,

khoih-prog, 2020-12-23:
You're so good and not a newbie. I'm slower than you many steps.

khoih-prog, 2020-12-23:
Your contribution has been noted in 

1. [**Note for Platform IO using ESP32 LittleFS**](https://github.com/khoih-prog/ESPAsync_WiFiManager#note-for-platform-io-using-esp32-littlefs)
2. [**Contributions and Thanks**](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks).

Regards,

solroshan, 2020-12-23:
Oh, my bad. I did not well look into your notes. I see, you wrote everything before bigbang :embarrassed:

Also, I feel proud to see my name in this clean work's [Contributions and Thanks](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks) section. :1st_place_medal: