khoih-prog/ESPAsync_WiFiManager #29: Help with an example.

mojgithub, 2021-01-03:
Hi,

This is my first try with this wifi manager. 

ESP32-S2 board. 
fresh arduino IDE 1.8.13
I have tested that board works.

Added your lib: ESPAsync_WiFiManager

Example: Async_AutoConnect.ino (no changes!)

Compile:
[image]

I have connected to AutoConnectAP.
I have opened tab with 192.168.100.1

My Serial:
[image]

`Starting Async_AutoConnectAP using LittleFS on ESP32S2_DEV
ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = AutoConnectAP
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = espressif, Pass = 
Got stored Credentials. Timeout 120s
We haven't got any access point credentials, so get them now
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] 
Configuring AP SSID = ESP_5A1DF7C_AutoConnectAP
[WM] AP PWD = MyESP_5A1DF7C
[WM] AP Channel = 7
[WM] Custom AP IP/GW/Subnet =  192.168.100.1 192.168.100.1 255.255.255.0
[WM] AP IP address = 192.168.100.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
`

My chrome inspector for 192.168.100.1
Most of the time Pending ...
[image]

Problem:

PC Windows 10
99% of the time i get white blank page on 192.168.100.1
1% i get menu page. If i am lucky with that i can click buttons but nothing happens ... i am redirected for example to 192.168.100.1/i  .. which stays white blank as any other page

Android: 100% i get white blank page

What i am doing wrong?

khoih-prog, 2021-01-03:
Thanks for your interest in the library.

> 99% of the time i get white blank page on 192.168.100.1

It's totally difficult to guess what's wrong in your scenario, I just can guess and suggest something for you to try and verify

1. Is there too many WiFi APs and **channel conflict** ? Any problem with Browser, Win10, Library/IDE installations?
2. You're using **ESP32-S2,** did you do anything special for this new ESP32 (which ESP32 core, Tools->Boards selected) As I currently don't have any S2 to test, I don't know if it's OK to use the `ESP32 core 1.0.4 without any touch`. **ESP32-S2 has only 1 core, instead of 2**, so I **don't think (and haven't verified)** current `ESP32 core 1.0.4 without any touch` will work without any WiFi issue.

[image]

3. Do you have any **classic and popular ESP32** (ESP32, ESP32-S) to test to isolate the issue (not channel conflict, Browser, Win10, Library/IDE installations, etc)
4. Use other examples such as [**ConfigOnDRD_ESP32_minimal**](https://github.com/khoih-prog/ESP_WiFiManager/tree/master/examples/ConfigOnDRD_ESP32_minimal) and [**Async_ConfigOnDoubleReset**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDoubleReset) to see if there is any difference.

Please share your experience, if it's working for you. If not, I might buy some ESP32-S2 to test to know what's happening. Then I can add ESP32-S2 into the supported list.

Good Luck,

alexmohr, 2021-01-03:
I have the exact same issue when running on esp8266 so I don't think it's due to the mcu. 
It seems like there is a buffer overflow or something like that. The html response will contain binary data.
Running `curl` will print a warning about that. 
The `page` string printed via serial seems correct.

khoih-prog, 2021-01-03:
@alexmohr 

Check your code if you modified from the examples. **Just one bit wrong or written to wrong place** (because of the code or bad hardware memory/flash) can cause lots of headache, and even days to methodologically find out where it is, if possible and lucky.

---

@mojgithub 

I just tested again on ESP32 and haven't seen anything wrong yet. The WebServer is working perfectly with Chrome and Ubuntu

---
---

#### Async_AutoConnectAP using LittleFS on NodeMCU_32S

```
Starting Async_AutoConnectAP using LittleFS on NodeMCU_32S
ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = AutoConnectAP
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
Got stored Credentials. Timeout 120s
[WM] LoadWiFiCfgFile 
[WM] failed
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -30
[WM] Channel: 2 ,IP address: 192.168.2.101
After waiting 4.03 secs more in setup(), connection result is connected. Local IP: 192.168.2.101
[WM] freeing allocated params!
HH
```
---

### Async_AutoConnectAP using LittleFS on ESP32_DEV

```
Starting Async_AutoConnectAP using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = AutoConnectAP
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
Got stored Credentials. Timeout 120s
[WM] LoadWiFiCfgFile 
[WM] failed
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -30
[WM] Channel: 2 ,IP address: 192.168.2.101
After waiting 4.43 secs more in setup(), connection result is connected. Local IP: 192.168.2.101
[WM] freeing allocated params!
H
```
---

#### Async_AutoConnectAP using LittleFS on ESP32_DEV

```
Starting Async_AutoConnectAP using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = AutoConnectAP
[WM] Previous settings invalidated
E (242) wifi: esp_wifi_disconnect 1153 wifi not start
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = , Pass = 
No stored Credentials. No timeout
We haven't got any access point credentials, so get them now
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] 
Configuring AP SSID = ESP_9ABF498_AutoConnectAP
[WM] AP PWD = MyESP_9ABF498
[WM] AP Channel = 5
[WM] Custom AP IP/GW/Subnet =  192.168.100.1 192.168.100.1 255.255.255.0
[WM] AP IP address = 192.168.100.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
dhcps: send_offer>>udp_sendto result 0
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Custom STA IP/GW/Subnet
[WM] DNS1 and DNS2 set
[WM] setWifiStaticIP IP = 192.168.2.232
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 2.60
[WM] Local ip = 192.168.2.232
[WM] Connection result:  WL_NO_SSID_AVAIL
[WM] Failed to connect
Not connected to WiFi but continuing anyway.
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
[WM] getSTAStaticIPConfig
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] SaveWiFiCfgFile 
[WM] OK
After waiting 0.00 secs more in setup(), connection result is WL_DISCONNECTED
[WM] freeing allocated params!

WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet2 ,RSSI= -49
[WM] Channel: 4 ,IP address: 192.168.2.232
HH
```

[image]

[image]

alexmohr, 2021-01-03:
As already mentioned the example is not modfied in anyway.
After running building everything from scratch `curl` seems to get the index page everything. 
Neither firefox nor chrome are able to display the page because it seems to be truncated. 

The information and wifi page is empty when regardless whether curl or a browser is used. 

I've set the following build parameters in `platformio.ini`
```
    -D_ESPASYNC_WIFIMGR_LOGLEVEL_=4
    -D DEBUG_ESP_PORT=Serial
```

What input can we provide to simplify debugging for you?

khoih-prog, 2021-01-03:
Are you two the same people. I'm confused that @alexmohr is different from @mojgithub.

If so, did you follow the suggested steps to isolate the issue? **I don't think I can help anything in your situation, especially with ESP32-S2.**

---
---

And this is from ESP8266 test.  No issue at all.

---

#### Async_AutoConnectAP using LittleFS on ESP8266_NODEMCU

```
Starting Async_AutoConnectAP using LittleFS on ESP8266_NODEMCU
ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = AutoConnectAP
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = HueNet1, Pass = 12345678
Got stored Credentials. Timeout 120s
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.188 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -39
[WM] Channel: 2 ,IP address: 192.168.2.188
After waiting 3.53 secs more in setup(), connection result is connected. Local IP: 192.168.2.188
[WM] freeing allocated params!
HHHHHHHHHH HHHH
```
---

#### Async_AutoConnectAP using LittleFS on ESP8266_NODEMCU

```
Starting Async_AutoConnectAP using LittleFS on ESP8266_NODEMCU
.....
[WM] setAPStaticIPConfig
[WM] setSTAStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
Stored: SSID = , Pass = 
No stored Credentials. No timeout
We haven't got any access point credentials, so get them now
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] 
Configuring AP SSID = ESP_702FF3_AutoConnectAP
[WM] AP PWD = MyESP_702FF3
[WM] AP Channel = 2
[WM] Custom AP IP/GW/Subnet =  192.168.100.1 192.168.100.1 255.255.255.0
[WM] AP IP address = 192.168.100.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
[WM] Custom STA IP/GW/Subnet
[WM] DNS1 and DNS2 set
[WM] setWifiStaticIP IP = 192.168.2.186
[WM] Connected after waiting (s) : 0.00
[WM] Local ip = 192.168.2.186
[WM] Timed out connection result: WL_IDLE_STATUS
Not connected to WiFi but continuing anyway.
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
[WM] getSTAStaticIPConfig
[WM] stationIP = 192.168.2.186 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] SaveWiFiCfgFile 
[WM] OK
After waiting 0.00 secs more in setup(), connection result is WL_IDLE_STATUS
[WM] freeing allocated params!

WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -32
[WM] Channel: 2 ,IP address: 192.168.2.186
HHHHHHH
```

[image]

[image]

alexmohr, 2021-01-03:
Two different persons, I hijacked the issue @mojgithub created... sorry for the confusion. 
As for the bug(?). I did a bit of digging through the library after i've changed `beginResponse` in the WifiManger to `beginResponseStream` (and added the necessary call to the stream print method) it seems to works flawless for me.

I'm still confused that it is working for you without any issues and probably I've done something wrong with the setup.
I'll create a new github repository to reproduce the issue and post the link here if that's okay for you

alexmohr, 2021-01-03:
https://github.com/alexmohr/ESPAsync_WiFiManager_debug

PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash

With the linked repository the index page is stable and access works every time. 
Accessing `/wifi` does not work properly.

```
curl http://192.168.100.1/wifi\? 
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
<!DOCTYPE html><html lang="en"><head><meta name="viewport" content="width=device-width, initial-scale=1, user-scalable=no"/><title>Config ESP</title><script>function c(l){document.getElementById('s').value=l.innerText||l.textContent;document.getElementById('p').focus();document.getElementById('s1').value=l.innerText||l.textContent;document.getElementById('p1').focus();}</script><script>(function(e){var t=function(){"use strict";var e="s",n=function(e){var t=-e.getTimezoneOffset();return t!==null?t:0},r=function(e,t,n){var r=new Date;return e!==undefined&&r.setFullYear(e),r.setDate(n),r.setMonth(t),r},i=function(e){return n(r(e,0,2))},s=function(e){return n(r(e,5,2))},o=function(e){var t=e.getMonth()>7?s(e.getFullYear()):i(e.getFullYear()),r=n(e);return t-r!==0},u=function(){var t=i(),n=s(),r=i()-s();return r<0?t+",1":r>0?n+",1,"+e:t+",0"},a=function(){var e=u();return ne%   
```

I will look more into migrating to `beginResponseStream` instead of `beginResponse` as this solution should have minimal impact on other mcu's and seems promising.
If you find something obviously wrong with the repo I've created please let me know.

khoih-prog, 2021-01-04:
First, you have to follow the Pre-requisites by updating the very-outdated core ESP8266 from v2.6.2 to v2.7.4 This might be the sole issue.

[image]

I don't like PIO much, and suggest you try **the examples with Arduino IDE to be sure it's working first before moving to PIO.
If there is any problem then, it's PIO issue.**

alexmohr, 2021-01-04:
I prefer using pio because it's much simpler and can be integrated in other IDEs. 

For the record if any one else also faces this issue: Disabling NTP via `USE_ESP_WIFIMANAGER_NTP=false` did the trick for me.

PS:
I don't think it's cool that you've already closed the issue because @mojgithub was the original author of it and his problem still seems to be unsolved and he might not be using PIO.

alexmohr, 2021-01-04:
To prevent confusion for future readers which might face the same issue platformio v2.6.2  is the latest version as of today which provides espcore in the version 2.7.4.

khoih-prog, 2021-01-04:
Thanks for clarifying. I forgot that PIO is using a different version from standard IDE

```
Platform espressif8266
--------
Updating platformio/espressif8266             2.6.2                              [Up-to-date]
Updating platformio/toolchain-xtensa          2.40802.200502 @ ~2.40802.191122   [Up-to-date]
Updating framework-arduinoespressif8266       3.20704.0 @ ~3.20704.0             [Up-to-date]
Updating platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]

```

mojgithub, 2021-01-04:
1. I live in different timezone as u so i cant reply immediately(sleep, work, etc).
2. **Anyway i have tried classic ESP32. WIN10, chrome it works.** But on Android, chrome i cant get menu. But i dont care about android if PC works.
3. So question is why ESP32-S2 doesnt work for me.

a) I have followed this instruction for installing ESP32-S2 board on Arduino IDE: https://www.mischianti.org/2020/12/01/esp32-s2-pinout-specs-and-arduino-ide-configuration-1/

b) I have tested it with 2 devices. 1st was DEV board esp32-S2 and 2nd was wroom esp32-s2. Wifi module on both devices work as i have tested it with different example scripts.

c) No there are no traffic on my channel. Only my main wifi router and esp32-s2. I have also shut down my main wifi router just to be sure. 

d) Arduino IDE is fresh and i use only your lib with dependency ...

e) I have tried few other examples from wifi manager and the same result.

Could u verify this with some other person? Or even yourself.

There is no question which unit i will use in future for my projects. It has to have low power consumption and lots of pins. 
_"ESP32-S2 is much more energy efficient than ESP32, in both CPU power consumption as well as RF power consumption. In some cases, ESP32-S2 is more than 50% more efficient than ESP32."_ 

So i would definitely like to use your lib in the future with ESP32-S2.

khoih-prog, 2021-01-04:
Now it's obvious that the **ESP32-S2 (board, package, etc.)** is still n**ot fully compatible yet with classic ESP32**, That's why it's not included in the stable branch of ESP32, and still requires manual and complex steps to add the ESP32-S2 package into Arduino IDE.

I've ordered some ESP32-S2 boards, and will start investigate, add the support to ESP32-S2 after receiving the new boards, if possible and the ESP32-S2 package is mature enough.

Please post any more **information and experience you think helpful** to save us some precious time to duplicate the research.

khoih-prog, 2021-01-04:
As you live in very different TZ without good NTP service, try to disable NTP by 

```
#define USE_ESP_WIFIMANAGER_NTP           false
#define USE_CLOUDFLARE_NTP                      false
```

to see if NTP / Cloudfare can cause some issue for you as some people have experienced.

mojgithub, 2021-01-04:
> As you live in very different TZ without good NTP service, try to disable NTP by
> 
> ```
> #define USE_ESP_WIFIMANAGER_NTP           false
> #define USE_CLOUDFLARE_NTP                false
> ```
> 
> to see if NTP / Cloudfare can cause some issue for you as some people have experienced.

It is already disabled in Async_AutoConnect.ino.

khoih-prog, 2021-01-04:
It's clear now more works have to be done to support ESP32-S2.

khoih-prog, 2021-02-13:
I'll publish a new release of [**ESPAsync_WiFiManager**](https://github.com/khoih-prog/ESPAsync_WiFiManager) to support ESP32-S2 within several days.

You can try now these libraries already supporting ESP32-S2

1. [**ESP_WiFiManager_Lite v1.1.0+**](https://github.com/khoih-prog/ESP_WiFiManager_Lite)
2. [**ESPAsync_WiFiManager_Lite v 1.1.0+**](https://github.com/khoih-prog/ESPAsync_WiFiManager_Lite)
3. [**ESP_WiFiManager v1.5.0+**](https://github.com/khoih-prog/ESP_WiFiManager)

khoih-prog, 2021-02-16:
### Major Releases v1.5.0

1. Add support to **ESP32-S2 (ESP32-S2 Saola, AI-Thinker ESP-12K, ESP32S2 Dev Module, UM FeatherS2, UM ProS2, microS2, etc.)**
2. Add [**Instructions to install ESP32-S2 core**](https://github.com/khoih-prog/ESPAsync_WiFiManager#howto-install-esp32-s2-core-for-esp32-s2-saola-ai-thinker-esp-12k-boards-into-arduino-ide)
3. Rewrite library code to be more efficient and multitask friendly

closed 2021-02-16 by khoih-prog