khoih-prog/ESPAsync_WiFiManager #33: ESP8266 Clear SSID and Pass

molillo, 2021-01-13:
Hi, good job!
I have a trouble, with ESPAsync_WiFiManager:
1.- Start Captive portal,  and I config SSID and Password for mi mobile wifi AP.
2.- Save data OK.
3.- manual Reset ESP8266 and start Config Portal for 30sg (I configure),   and next connect with my mobile AP perfect.
4.-  If I switch off  my mobile AP,  and reset ESP8266,  then:

**ESPAsync_WiFiManager Version v1.3.0
[WM] RFC925 Hostname = ConfigOnStartup
Stored: SSID = , Pass = 
Opening configuration portal.Open Config Portal without Timeout: No stored Credentials**

And no Stored SSID and no Pass !!!

Can you help me ?  
Thanks

molillo, 2021-01-13:
This is the log:

**INIT BEFORE FLASH ESP8266:**
Starting Async_ConfigOnStartup with DoubleResetDetect using LittleFS on ESP8266_GENERIC
ESPAsync_WiFiManager Version ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = ConfigOnStartup
Stored: SSID = , Pass = 
Opening configuration portal.Open Config Portal without Timeout: No stored Credentials.
Starting configuration portal.
[WM] WiFi.waitForConnectResult Done
[WM] SET AP
[WM] 
Configuring AP SSID = VISORCO2_FD6DB1
[WM] AP PWD = 12345678
[WM] AP Channel = 2
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Can't use Custom STA IP/GW/Subnet
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 3.35
[WM] Local ip = 192.168.43.254
[WM] Connection result:  WL_CONNECTED
WiFi connected...yeey :)
[WM] * Add SSID =  1-Xiaomi MCM , PW =  658864793Aa
[WM] SaveWiFiCfgFile 
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.43.254
[WM] freeing allocated params!
HH

**I SWITCH OFF THE MOBILE WIFI,  AND SWITCH ON:**
WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Additional SSID =  1-Xiaomi MCM , PW =  658864793Aa
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: 1-Xiaomi MCM ,RSSI= -61
[WM] Channel: 2 ,IP address: 192.168.43.254
H

**I SWITCH OFF THE MOBILE WIFI,  AND  BUTTON RESET IN ESP8266**
Starting Async_ConfigOnStartup with DoubleResetDetect using LittleFS on ESP8266_GENERIC
ESPAsync_WiFiManager Version ESPAsync_WiFiManager v1.4.1
[WM] RFC925 Hostname = ConfigOnStartup
_Stored: SSID = , Pass =_ 
Opening configuration portal.Open Config Portal without Timeout: No stored Credentials.

This is my problem !!   If I switch on ESP8266 with credentials saved,  and not find my mobile AP, then No Store Credentials ... 

thanks

khoih-prog, 2021-01-14:
Thanks for your clear and detailed bug report, the issue has been fixed in v1.4.2, with a note about your contribution in [**Contributions and Thanks**](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks).

Please test and verify if everything is OK now.

Regards,

---

9. Thanks to [**Manuel Capilla**](https://github.com/molillo) for reporting [ESP8266 Clear SSID and Pass](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/33) bug which is fixed and leading to v1.4.2.
---

### Releases v1.4.2

1. **Fix examples' bug not using saved WiFi Credentials after losing all WiFi connections.**
2. Fix compiler warnings.

---

closed 2021-01-14 by khoih-prog