khoih-prog/ESPAsync_WiFiManager #34: VERSION 1.4.2  NO MULTIWIFI

molillo, 2021-01-14:
Hi, good job !!   The las Issue in 1.4.1  it´s now Ok,  but ... ...
At the moment, not remember stored SSID and Pass,  and only use the last  pair  used for connection.
I use the   Async_ConfigOnStartup.ino 1.4.2,   and change     #define NUM_WIFI_CREDENTIALS      from 2 to 4

My ESP8266 connect and remember perfectly my mobile AP,  in all the test,   but  if I   switch off my mobile AP,  and config SSID and Password for use my fixed router AP,  ESP8266 remember only my fixed AP,  and not connect with mobile AP. (don´t store data)

thanks

khoih-prog, 2021-01-14:
It's not working that way.

The **NUM_WIFI_CREDENTIALS is currently fixed at 2**, and even you increased it to larger numbers, the library still stores maximum 2 WiFi AP Credentials.

molillo, 2021-01-14:
Hi,  I change NUM_WIFI_CREDENTIALS to 2,  and reflash all, cleaning wifi data. The problem it´s:

I configure portal ESP8266  my mobile WiFI SSID and password:
ESPAsync_WiFiManager v1.4.2
[WM] RFC925 Hostname = ConfigOnStartup
[WM] setAPStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP **Self-Stored: SSID = 1-Xiaomi MCM, Pass = 658864793Aa**
Opening configuration portal.
[WM] * **Add SSID =  1-Xiaomi MCM , PW =  658864793Aa**
Got ESP Self-Stored Credentials. Timeout 60s for Config Portal
Starting configuration portal.
[WM] WiFi.waitForConnectResult Done
[WM] SET AP_STA
[WM] 
Configuring AP SSID = VISORCO2_FD6DB1
[WM] AP PWD = 12345678
[WM] AP Channel = 4
[WM] Custom AP IP/GW/Subnet =  192.168.4.1 192.168.4.1 255.255.255.0
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] ESPAsync_WiFiManager::startConfigPortal : Enter loop
[WM] Connecting to new AP
[WM] Previous settings invalidated
[WM] Can't use Custom STA IP/GW/Subnet
[WM] Connect to new WiFi using new IP parameters
[WM] Connected after waiting (s) : 8.41
[WM] Local ip = 192.168.1.138
[WM] Connection result:  WL_CONNECTED
WiFi connected...yeey :)
**At the moment,  it´s ok**

I restart my ESP8266, and configure in portal new router wifi:

[WM] * **Add SSID =  DIGIFIBRA-pUNq , PW =  DAYhpNt6KK**
[WM] getSTAStaticIPConfig
[WM] stationIP = (IP unset) , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] SaveWiFiCfgFile 
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.1.138
[WM] freeing allocated params!
HH{l

I restart my ESP8266, and switch off, my ROUTER WIFI.  then must remember my mobile AP, but not,  It´s awaiting for ROUTER WIFI:

Starting Async_ConfigOnStartup with DoubleResetDetect using LittleFS on ESP8266_GENERIC
ESPAsync_WiFiManager v1.4.2
[WM] RFC925 Hostname = ConfigOnStartup
[WM] setAPStaticIPConfig
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
E**SP Self-Stored: SSID = DIGIFIBRA-pUNq, Pass = DAYhpNt6KK**
Opening configuration portal.
[WM] * **Add SSID =  DIGIFIBRA-pUNq , PW =  DAYhpNt6KK**
Got ESP Self-Stored Credentials. Timeout 60s for Config Portal
Starting configuration portal.
[WM] WiFi.waitForConnectResult Done
[WM] SET AP_STA
[WM] 

Self-Stored: SSID ... ...    and Add SSID ... ...    IT´S  repeat the wifi router,  and don´t remember my mobile AP.

Thanks

khoih-prog, 2021-01-14:
You have to **add both AP SSIDs in Config Portal to be saved.**

The so-called `ESP Self-Stored: SSID = DIGIFIBRA-pUNq, Pass = 12345678` is not persistent, and is temporarily stored in Flash by ESP, and will be erased whenever you either connect to new AP or lose connection. That's why you'll lose that info.

You have to see in the Debug Terminal something like

```
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet2 , Router_Pass =  password2
[WM] * Additional SSID =  HueNet1 , PW =  password1
[WM] * Additional SSID =  HueNet2 , PW =  password1
```

to know both Credentials are stored in FileFS/EEPROM.

I suggest you you this [**Async_ConfigOnDoubleReset example**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDoubleReset) to enter the Portal more easily by DoubleReset to verify.

closed 2021-01-14 by khoih-prog