khoih-prog/ESPAsync_WiFiManager #35: Captive portal is not triggered on IOS

reneklootwijk, 2021-01-15:
On my iPhone running IOS version 14.3 the captive portal is not triggered. When I connect to the AP the first time I have to explicitly go to 192.168.100.1 in a browser to get to the config screen. I tried other frameworks, e.g. AutoConnect, and there the captive portal is automatically triggered.

haydenmkoch, 2021-01-15:
Try without setting a static IP for the AP I.e remove `ESPAsync_wifiManager.setAPStaticIPConfig(...)` This solved the issue for me and the captive portal now triggers on the default 192.168.4.1

khoih-prog, 2021-01-16:
@haydenmkoch 

Thanks for helping here. You're absolutely correct.

The iPhone (or a WebBrowser on any device) will auto-redirect to the Captive Portal IP `192.168.4.1`, which is a default AP IP set by ESP (private-code) cores. Currently, the only way Captive Portal is auto-triggered and opened is to use that `192.168.4.1` AP IP. Otherwise, you have to manually open the CP.

closed 2021-01-16 by khoih-prog