khoih-prog/ESPAsync_WiFiManager #40: Feature request: Provide example using espidf framework

jcwren, 2021-03-05:
All the examples provided are using the `arduino` framework.

An example using the `espidf` framework would be desirable, particularly a non-C++ example.

According to the GitHub tags and the `platformio/platformio.ini` file, the `espidf` framework is supported for the `espressif32` platform. If this actually isn't the case, perhaps those tags should be removed?

khoih-prog, 2021-03-05:
Sorry I have no time to do that. If you are using the different platform / framework, you already have good knowledge and have to do the conversion yourself.

closed 2021-03-05 by khoih-prog

safinea, 2021-07-03:
To be fair, I think @jcwren has a point... I am also struggling to get this library working with the `espidf` framework, which this library says it supports. One example of a problem seems to be that `ESPAsync_WiFiManager.h` tries to include `WiFi.h` (line 103 in v.1.9.1) which exists only in the `arduino` framework (as far as I can tell). There's a suggestion it will work, but no example and the source code _seems_ to not support it. Any help is appreciated.

khoih-prog, 2021-07-03:
Thanks for your interest in the library.

Actually, the `espidf` framework is not intentionally supported, and the `espidf` in `platformio.ini` is just a mistaken left-over when the file was copied from other source.

The library is written and intended to support only **Arduino** users, normally with less experience than PRO people using other platforms, such as plain `espidf` or `platformio`. 

If the other platform can use the library, it's just **unintended plus**.

I always trust the PRO people can **simply adjust the library** according to their current needs, and for efficiency, I'd also rather spend time to help a larger number and more people than just a very limited / small quantity of people.

Hopefully, you'll easily solve the issue soon as this library /  ESP32 / Arduino code, all has to based on `espidf`. At that time, I'd appreciate it if you create a PR to inform and help other users.

Sorry and Good Luck,

khoih-prog, 2021-07-05:
You can use this [**esp32-wifi-manager**](https://github.com/tonyp7/esp32-wifi-manager) written specifically for `espidf`.

safinea, 2021-07-05:
Hi Khoi,

 

Thanks so much for your open an honest answer… It’s much appreciated. As is the link to an espidf version.

 

I’m no noob (embedded sw background) but no pro either (esp32 multicore is new), so I’m still contemplating whether to keep my projet in espidf, of move over to Arduino framework (especially since I understand that that also still allows me to use FreeRTOS… 😊)

 

Again, thank you for your efforts in creating and maintaining this lib and for your feedback!

 

Best regards, Marcel van der Laan

 

 <https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 	Scanned by McAfee <https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient>  and confirmed virus-free.

khoih-prog, 2021-07-05:
Hi,

> I’m no noob (embedded sw background) but no pro either

We all are

> move over to Arduino framework

IMHO, this is much better as you can `stand on other people's shoulders` to faster and easier have higher reach.

I've done much more complicated projects, using only **Arduino** platform. Certainly in Embedded SW engineering (and almost anything), **problem solving skills**, good design, passion and knowledge are more important than **fancy platforms** (with lot more bugs and issues without real benefits)