khoih-prog/ESPAsync_WiFiManager #43: SyncClient.cpp: fatal error: interrupts.h: No such file

olmiv, 2021-03-27:
...\Arduino\libraries\ESPAsyncWebServer-master\src\SyncClient.cpp:25:24: fatal error: interrupts.h: No such file or directory

Hello. Help me understand what's going on. All your websocket-related projects have stopped compiling in the Arduino IDE for ESP 32. The same error always appears:
"SyncClient.cpp: fatal error: interrupt. h: No such file..."
My project with your libraries stopped compiling because of this error. I tried to take your example "Async_ESP 32_Webserver", which my project is based on, but compiling your example also ends up with the same error. I tried reinstalling the Arduino IDE, but the result is the same. What could be the problem. The request goes to the ESP8266 library, right? What and where to fix, so that the compiler does not access this library. I read that this bug is fixed in platformio, but I'm not a programmer and it's a complicated thing for me. Thank you in advance.

khoih-prog, 2021-03-27:
Thanks for your interests in the library.

I'm sorry that you got into an issue which is **not caused by this library,** but possibly from your installations and/or version mismatching. Reinstalling Arduino IDE and libraries to see if the issue gone.

A quick [**Google search**](https://www.google.com/search?q=SyncClient.cpp%3A+fatal+error%3A+interrupt.+h%3A+No+such+file...&oq=SyncClient.cpp%3A+fatal+error%3A+interrupt.+h%3A+No+such+file...&aqs=chrome..69i57.1278j0j15&sourceid=chrome&ie=UTF-8) will give you some clues about what happening and how people have solved it.

I'm sorry I can't help any further and suggest you to search for help in ESP32 or Arduino forum. I'm closing the issue now.

Good Luck with your project,

closed 2021-03-27 by khoih-prog

olmiv, 2021-03-28:
I'm afraid you're wrong. I found the library "ESPAsyncWebServer" from the beginning of 2020, there is no file in it SyncClient.cpp. With this library, there are no problems with compiling for ESP32 in the Arduino IDE. Perhaps the error is in this file?

khoih-prog, 2021-03-28:
Something is very wrong with your installations / settings and only you can find out and solve the issue. I won't waste my time anymore here.

---

At least you have to verify why 

> compiling in the Arduino IDE for ESP 32

and 

> The request goes to the ESP8266 library

because `SyncClient.cpp/h` are only for [**ESP8266's ESPAsyncTCP**](https://github.com/me-no-dev/ESPAsyncTCP)

Check if you have latest ESP32 core version, at least v1.0.4-v1.0.6

From [**ESPAsyncWebServer**](https://github.com/me-no-dev/ESPAsyncWebServer#espasyncwebserver)

> For ESP8266 it requires ESPAsyncTCP To use this library you might need to have the latest git versions of ESP8266 Arduino Core
> For ESP32 it requires AsyncTCP to work To use this library you might need to have the latest git versions of ESP32 Arduino Core