khoih-prog/ESPAsync_WiFiManager #44: Can I use this?

dduehren, 2021-03-31:
Using a library should be simple, but this thing has gotten so complicated with the number of configuration variables, etc that it's hard to see the forest for the trees.

What I want to do is to add something like wifimanager to my existing app  where I have a 9 page asyc web server with jquery and style sheets, etc, with my own NTP time server, etc.  I found it much easier to start with a simple auto-connect to my local LAN for development.   But I want the product to be user configured for his/her local network and/or for it to work in stand-along AP mode.

From what I've read it looks like what I need to do is add another IP address for my site, but I think yours may be overly complicated.

So what I need:
1. Start in AP mode, display available networks and capture login info.  Try to connect and remember the connection, possibly adding to other connections.
2. If logged in, switch to my webserver.
3. Provide an option in AP mode to bypass the login stuff, and operate stand-alone, for example, if not in range of a LAN., but also connect to my web server.
4.  There's a lot of presumption in your code about file systems, etc.  I already have SPIFFS set up, And as I mentioned, NTP.  You are not the center of the universe, just the wifi configuration manager.  I think it would be better if you had separate examples for SPIFF and LF than all the ifdefs etc. 
5. It also has to operate on cell-phones, why do I have to specifically configure for this?

dduehren, 2021-03-31:
> I'm sorry this generic library doesn't fit your complicated and customized purpose.
>
> I suggest that you hire some professional firm/people to write and customize for your application sand use cases.
>
> Good Luck,

Khoi,

 

Maybe I’ll adapt what I developed along these lines for another project of mine.  The effort is probably about the same as figuring out all the config options of your library.  I know this sounds brash and I’m sure you’ve put a lot of effort into that library.  And I’m sure you know what scope creep is.  You might want to stand back and look at what you’re offering.

 

*	David

khoih-prog, 2021-03-31:
Hi David,

I'd appreciate it if you can add features into the library by making PRs, or even better if you can make some new library with new features. 

A generic library has to take care of a wider range of users' requests / use cases and certainly is more complicated to configure and adapt. You have to understand that there is not a simple yet one-size-fits-all library.

I certainly prefer the UNIX-philosophy : **simple then chain-able**. But in the Arduino world, some users are not very sophisticated and request more features to save them time to write codes. That's the trade-off I had to make, but there is also some limits I don't like to pass if not useful to many more users.

As you also know it's so much a single guy can do with limited time. And I certainly have no time and interest to look at your specific use cases to understand.

Regards,