khoih-prog/ESPAsync_WiFiManager #47: Minor: examples/Async_ESP32_FSWebServer/ wrongly uses FileFS.begin(true)

robcazzaro, 2021-04-08:
(please also see https://github.com/lorol/LITTLEFS/issues/27 for more context)  

I'm using your library and your examples as a starting point for my project. Thanks for sharing this!

I noticed that the following code
```
// Format SPIFFS if not yet
  if (!FileFS.begin(true))
  {
    Serial.print(FS_Name);
    Serial.println(F(" failed! AutoFormatting."));
  }
```
seems to assume that FileFS.begin(true) returns false if the SPIFFS was un-initialized and needed to be formatted. In reality that call returns _true_ if the mount has been successful, even if that required an auto-format, _false_ only when the mount fails for whatever reason.

If the debug output is enabled, the LittleFS library debug messages overwrite any debug print from the Arduino thread, causing additional comprehension problems

If the call FileFS.begin(true) fails, that means that there is no SPIFFS, and your example should abort

The above applies to all your examples using SPIFFS for ESP32

khoih-prog, 2021-04-08:
Thanks for identifying the misleading message in the examples. It happened because time spent in them were so small, and not considered so serious.

I have been aware of this minor issue and knew it's better to have something like the following code, such as in [**BlynkSimpleEsp32_WM.h**#L1510-L1520](https://github.com/khoih-prog/Blynk_WM/blob/master/src/BlynkSimpleEsp32_WM.h#L1510-L1520) or [**BlynkSimpleEsp8266_WM.h**#L1467-L1480](https://github.com/khoih-prog/Blynk_WM/blob/master/src/**BlynkSimpleEsp8266_WM.h**#L1467-L1480)

I won't let it slipping this time and will fix it in next release.

1. For ESP32

```
     // Format SPIFFS if not yet
      if (!FileFS.begin(true))
      {
        Serial.println(F("SPIFFS/LittleFS failed! Already tried formatting."));
        
        if (!FileFS.begin())
        {
          Serial.println(F("SPIFFS/LittleFS failed! Pls use EEPROM."));
          return false;
        }
      }
```

2. For ESP8266

```
      if (!FileFS.begin())
      {
        FileFS.format();
        
        if (!FileFS.begin())
        {
#if USE_LITTLEFS
          Serial.println(F("LittleFS failed!. Please use SPIFFS or EEPROM."));
#else
          Serial.println(F("SPIFFS failed!. Please use LittleFS or EEPROM."));
#endif 
          return false;
        }
      }
```

3. Combined, to be in new release

```
  // Format FileFS if not yet
#ifdef ESP32
  if (!FileFS.begin(true))
#else
  if (!FileFS.begin())
#endif
  {
#ifdef ESP8266
    FileFS.format();
#endif

    Serial.println(F("SPIFFS/LittleFS failed! Already tried formatting."));
  
    if (!FileFS.begin())
    {     
#if USE_LITTLEFS
      Serial.println(F("LittleFS failed!. Please use SPIFFS or EEPROM. Stay forever"));
#else
      Serial.println(F("SPIFFS failed!. Please use LittleFS or EEPROM. Stay forever"));
#endif

      while (true)
      {
        delay(1);
      }
    }
  }
```

robcazzaro, 2021-04-09:
I know it's not really a meaningful contribution, but I'm glad to have been of help :)

Your new code looks great to me. I cannot thank you enough for contributing this library and the examples. It really helped me a lot, and I learned a lot just following the examples

Just keep in mind that, if using LittleFS at least, the debug messages from LittleFS will preempt your Serial.print() and the user won't see a debug message, just the code hanging. If you do something like the code below, on the other hand, the error message will be correctly displayed. Truly minor, but it took me a bit to figure out what was happening, and remembering that all ESP32 code is dual thread, unlike other Arduino platforms

```
  // Format FileFS if not yet
#ifdef ESP32
  if (!FileFS.begin(true))
#else
  if (!FileFS.begin())
#endif
  {
#ifdef ESP8266
    FileFS.format();
#endif

    Serial.println(F("SPIFFS/LittleFS failed! Already tried formatting."));
  
    if (!FileFS.begin())
    {     
       delay(100); // prevents debug info from the library to hide err message. The code stops anyway, so a delay is not an issue
#if USE_LITTLEFS
      Serial.println(F("LittleFS failed!. Please use SPIFFS or EEPROM. Stay forever"));
#else
      Serial.println(F("SPIFFS failed!. Please use LittleFS or EEPROM. Stay forever"));
#endif

      while (true)
      {
        delay(1);
      }
    }
  }
```

khoih-prog, 2021-04-09:
That's great. I'll certainly use it and have some not about your contribution.
Regards,

khoih-prog, 2021-04-09:
The new [**ESPAsync_WiFiManager Release v1.6.2**](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.6.2) has been published, with a note to thank you at [**Contributions and Thanks**](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

---

### Releases v1.6.2

1. Fix example misleading messages. Check [**Minor: examples/Async_ESP32_FSWebServer/ wrongly uses FileFS.begin(true)** #47](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/47)

closed 2021-04-09 by khoih-prog