khoih-prog/ESPAsync_WiFiManager #48: use AsyncElegantOTA

jduisterwinkel, 2021-04-09:
Hi,
I used AsyncElegantOTA (https://github.com/ayushsharma82/AsyncElegantOTA) successfully in other projects, but in combination with ESPAsync_WiFiManager it won't compile.
I started from your example Async_ESP_FSWebServer_DRD.ino
Normally 3 lines of code are sufficient to setup AsyncElegantOTA in the code:
1. at the top: #include <AsyncElegantOTA.h>
2. just before line server.begin(); in setup(): AsyncElegantOTA.begin(&server);
3. somewhere in loop(): AsyncElegantOTA.loop(); 
Now after applying these 3 lines to your example the AsyncElegantOTA type is not recognized: 
> sketch_apr07a:229:1: error: 'AsyncElegantOTA' does not name a type
> AsyncElegantOTA.begin(&server);    // Start ElegantOTA (reach out to http://{ip-address}/update)
> ^
Is this my fault? Or can you explain this behavior?

khoih-prog, 2021-04-10:
Sorry I don't like to spend time on the **problem not caused from this library's bug.**

But I tested and compiled OK with **AsyncElegantOTA**.

[image]

You have to check your installations and only you can know and fix the issue.

Good Luck,

closed 2021-04-10 by khoih-prog

jduisterwinkel, 2021-04-10:
It was my fault, unfortunately. I putted the #include not on top of the code. Sorry for the inconveniance with this issue.