khoih-prog/ESPAsync_WiFiManager #50: Portal configuration window not opening

glen361, 2021-04-29:
I have been using the ESPAsync_WiFiManager 1.7.1 library on a ESP32 without issues. I then tried using the library on a Wemos D1, the portal opens, information page loads, but the configuration (wifi selection) page is blank. 

I tried several different examples from the GitHub. Tried accessing the Portal from another computer, on 192,168.4.1 getting the same result, a blank  Configuration page. I tried your ESP_WifiManager and the code/library works. 

Boards using:
esp32doit-devkit-v1
d1_mini

Visual Code Studio
Version: 1.55.2
Commit: 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date: 2021-04-13T09:36:32.643Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 20.3.0

Thank you for your help.
Glen

mruettgers, 2021-04-29:
Hi,

First of all, thank you for your great work on this library.

I'm encountering the same issue here with v1.7.0. It looks like the web server is returning garbage after some valid bytes (see screenshot attached).

The garbage occurrs after:
```<!DOCTYPE html><html lang="en"><head><meta name="viewport" content="width=device-width, initial-scale=1, user-scalable=no"/><title>Config ESP</title><script>function c(l){document.getElementById('s').value=l.innerText||l.textContent;document.getElementById('p').focus();document.getElementById('s1').value=l.innerText||l.textContent;document.getElementById('p1').focus();}</script><script>(function(e){var t=function(){"use strict";var e="s",n=function(e){var t=-e.getTimezoneOffset();return t!==null?t:0},r=function(e,t,n){var r=new Date;return e!==undefined&&r.setFullYear(e),r.setDate(n),r.setMonth(t),r},i=function(e){return n(r(e,0,2))},s=function(e){return n(r(e,5,2))},o=function(e){var t=e.getMonth()>7?s(e.getFullYear()):i(e.getFullYear()),r=n(e);return t-r!==0},u=function(){var t=i(),n=s(),r=i()-s();return r<0?t+",1":r>0?n+",1,"+e:t+",0"},a=function(){var e=u();return ne```

[image]

Thanks,
Michael

mruettgers, 2021-04-29:
Hm. It looks like the issue is caused by passing the `page` variable of type `String` by value.

Replacing 

```AsyncWebServerResponse *response = request->beginResponse(200, WM_HTTP_HEAD_CT, page);```

with 

```AsyncWebServerResponse *response = request->beginResponse_P(200, WM_HTTP_HEAD_CT, page.c_str());```

solves the issue for me.
And because there's no need to do memory operations for building the response, we should always use ` beginResponse_P`  here and pass the char pointer.
Unfortunately I don't have the time for a pull request right now, maybe within the next days.

khoih-prog, 2021-04-29:
@glen361 

Thanks for using the library.
I've retried many times, using WeMos D1 Mini as well as WeMos D1, ESP3288, etc. but everything is OK.

```
Starting Async_ConfigOnDoubleReset using LittleFS on ESP8266_WEMOS_D1MINI
ESPAsync_WiFiManager v1.7.1
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet, Pass = password
[WM] * Add SSID =  HueNet , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D01234
doubleResetDetected
Saving config file...
Saving config file OK
Open Config Portal without Timeout: Double Reset Detected
Starting configuration portal @ 192.168.4.1, SSID = ESP_AB1481, PWD = MyESP_AB1481
[WM] Connecting to new AP
WiFi connected...yeey :)
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
[WM] SaveWiFiCfgFile 
[WM] stationIP = 192.168.2.186 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.186
HHH
```

I think there is something wrong either with your board, your code or your development platform (Visual Code Studio). I suggest you to eliminate the issue one-by-one, by using the standard Arduino IDE v1.8.13 on another  ESP8266 board.

You have to know ESP32 is more powerful with larger resource than ESP8266. Therefore code, running on ESP32, can't always run on ESP8266.

I have to close the issue until you prove there is some bug in the library.

Good Luck,

closed 2021-04-29 by khoih-prog

glen361, 2021-04-29:
Ok thank you for your prompt response. I’ll give that a try. 

Glen

Closed #50.

khoih-prog, 2021-04-29:
@mruettgers 

I don't think that is the issue. If you have a look at the [ESPAsyncWebServer library #L246-L254](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L246-L254)

```
AsyncWebServerResponse *beginResponse(int code, const String& contentType=String(), const String& content=String());

...

AsyncWebServerResponse *beginResponse_P(int code, const String& contentType, PGM_P content, AwsTemplateProcessor callback=nullptr);
```

The following function call is correct, and String is passed by `reference`, not by value (with duplication) as you think.

```
AsyncWebServerResponse *response = request->beginResponse(200, WM_HTTP_HEAD_CT, page);
```

The following call to use `beginResponse_P` is wrong, unless your `page` is PGM_P type (in PROGMEM)

```
AsyncWebServerResponse *response = request->beginResponse_P(200, WM_HTTP_HEAD_CT, page.c_str());
```

I suggest you to check your code as well as your usage of your String for **heap / stack corruption** issue.