khoih-prog/ESPAsync_WiFiManager #51: How to retrieve timezone?

benpeart, 2021-04-30:
I have seen in the config web page that the time zone is detected and shown but I can't figure out how to retrieve that in my program so that I can use it to initialize a NTP SyncProvider for the time library I'm currently using (https://github.com/PaulStoffregen/Time).

khoih-prog, 2021-04-30:
Hi,

I already add the new requested feature and will publish new release within tomorrow.

Hereafter is the debug terminal

```
Starting Async_ConfigOnDoubleReset using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
Current Timezone is America/New_York        <===============  The timezone Name
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -30
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.94 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
HHHHHHHHHH HHH
```

benpeart, 2021-04-30:
Awesome. Thank you. That was fast!

I'm assuming I will need to read the stored value from the config file and am hoping it will store the offset from GMT as that is what I need to convert the NTP time to local time.

khoih-prog, 2021-05-01:
Will publish new **v1.8.0** soon, waiting for final touch.
The new feature will provide the complex TZ ( for example `TZ =  EST5EDT,M3.2.0,M11.1.0 `) so that you can use directly in ESP32/ESP8266 code as follows

```
#if ESP8266
      configTime(ESPAsync_wifiManager.getTZ(WM_config.TZ_Name), "pool.ntp.org"); 
#else
      //configTzTime(WM_config.TZ, "pool.ntp.org" );
      configTzTime(WM_config.TZ, "time.nist.gov", "0.pool.ntp.org", "1.pool.ntp.org");
#endif
```

The debug terminal outputs

### 1. ESP8266

```
Starting Async_ConfigOnDoubleReset_TZ using LittleFS on ESP8266_NODEMCU
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.188 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
[WM] Current TZ_Name = America/Toronto , TZ =  EST5EDT,M3.2.0,M11.1.0   <======= TZ name and TZ here
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -42
[WM] Channel: 2 ,IP address: 192.168.2.188
After waiting 3.49 secs more in setup(), connection result is connected. Local IP: 192.168.2.188
Stop doubleResetDetecting
Saving config file...
Saving config file OK
Local Date/Time: Sat May  1 12:34:50 2021
Local Date/Time: Sat May  1 12:35:50 2021
Local Date/Time: Sat May  1 12:36:50 2021
Local Date/Time: Sat May  1 12:37:50 2021
Local Date/Time: Sat May  1 12:38:50 2021
```

### 2. ESP32

```
Starting Async_ConfigOnDoubleReset_TZ using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.8.0
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
[WM] Current TZ_Name = America/New_York , TZ =  EST5EDT,M3.2.0,M11.1.0   <======= TZ name and TZ here
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet2 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet2 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -40
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.95 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
Local Date/Time: Sat May  1 11:52:50 2021
Local Date/Time: Sat May  1 11:53:50 2021
Local Date/Time: Sat May  1 11:54:50 2021
Local Date/Time: Sat May  1 11:55:50 2021
Local Date/Time: Sat May  1 11:56:50 2021
```

khoih-prog, 2021-05-02:
The [**ESPAsync_WiFiManager** Releases v1.8.0](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.8.0) has just been published.

Your request for enhancement, leading to new v1.8.0 has also been noted in [Contributions and Thanks](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

---

### Major Releases v1.8.0

1. Add auto-Timezone feature with variable `_timezoneName` (e.g. `America/New_York`) and function to retrieve TZ (e.g. `EST5EDT,M3.2.0,M11.1.0`) to use directly to configure ESP32/ESP8266 timezone. Check [How to retrieve timezone? #51](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/51)
2. Store those `_timezoneName` and `TZ` in LittleFS or SPIFFS config file.
3. Using these new timezone feature is optional.
4. Add checksum in config file to validate data read from LittleFS or SPIFFS config file.
5. Update examples to show how to use the new TZ feature.

closed 2021-05-02 by khoih-prog