khoih-prog/ESPAsync_WiFiManager #54: Call to ESPAsync_wifiManager.getTimezoneName() only works during initial config

benpeart, 2021-05-13:
I'm trying to use the new ESPAsync_wifiManager.getTimezoneName() function to set the timezone via configTzTime() on every boot. In my testing it only returns a result when I am in the initial config code path.

For example, if I trigger the initial config path by calling ESPAsync_wifiManager.resetSettings(), my call to ESPAsync_wifiManager.getTimezoneName(); correctly returns my time zone. If I just do a 'normal' boot, it returns an empty string.

I looked at the 20+ samples and all of them only call getTimezoneName when they are in their initialConfig state as well. Is this a known limitation? It seems like you should be able to set the timezone on 'regular' boots as well.

I may just be calling things incorrectly. My code is based on the Async_AutoConnect_ESP32_minimal sample - how should I be modifying it to correctly get the timezone?

```
  // connect to wifi or enter AP mode so it can be configured
  DB_PRINT("\nStarting Kaleidoscope on " + String(ARDUINO_BOARD));
  DB_PRINTLN(ESP_ASYNC_WIFIMANAGER_VERSION);
  ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "Kaleidoscope");
  //ESPAsync_wifiManager.resetSettings();   //reset saved settings
  ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 132, 1), IPAddress(192, 168, 132, 1), IPAddress(255, 255, 255, 0));
  ESPAsync_wifiManager.autoConnect("KaleidoscopeAP");
  if (WiFi.status() == WL_CONNECTED)
  {
    DB_PRINT(F("Connected. Local IP: "));
    DB_PRINTLN(WiFi.localIP());
  }
  else
  {
    DB_PRINTLN(ESPAsync_wifiManager.getStatus(WiFi.status()));
  }
  String tempTZ = ESPAsync_wifiManager.getTimezoneName();
  if (tempTZ.length())
    configTzTime(tempTZ.c_str(), "us.pool.ntp.org", "time.nist.gov", "0.pool.ntp.org");
  DB_PRINTF("Current timezone is %s\r\n", tempTZ.c_str());

```

khoih-prog, 2021-05-13:
That function is used for internal work only.

The TZ-related data are stored in WM_config and you have to accessed from there

```
typedef struct
{
  WiFi_Credentials  WiFi_Creds [NUM_WIFI_CREDENTIALS];
  char TZ_Name[TZNAME_MAX_LEN];     // "America/Toronto"
  char TZ[TIMEZONE_MAX_LEN];        // "EST5EDT,M3.2.0,M11.1.0"
  uint16_t checksum;
} WM_Config;

WM_Config         WM_config;

...

#if USE_ESP_WIFIMANAGER_NTP      
    if ( strlen(WM_config.TZ_Name) > 0 )
    {
      LOGERROR3(F("Current TZ_Name ="), WM_config.TZ_Name, F(", TZ = "), WM_config.TZ);

  #if ESP8266
      configTime(WM_config.TZ, "pool.ntp.org"); 
  #else
      //configTzTime(WM_config.TZ, "pool.ntp.org" );
      configTzTime(WM_config.TZ, "time.nist.gov", "0.pool.ntp.org", "1.pool.ntp.org");
  #endif   
    }
    else
    {
      Serial.println(F("Current Timezone is not set. Enter Config Portal to set."));
    } 
#endif
```

closed 2021-05-13 by khoih-prog

benpeart, 2021-05-13:
What is the recommended way to get the WM_Config structure loaded with the persisted config? I see about 300 lines of code in setup() copy/pasted into most of the samples that includes a call to loadConfigData() but when I tried to extract the ~100 lines that seemed needed to load the config structure - I get an error:

```
.pio\libdeps\node32s\LittleFS_esp32\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
E (9060) esp_littlefs: mount failed,  (-84)
E (9064) esp_littlefs: Failed to initialize LittleFS
[E][LITTLEFS.cpp:90] begin(): Mounting LITTLEFS failed! Error: -1
```

This setup code is also the same code that contains the call to getTimezoneName() that you said is internal only so I'm unsure how much of it I should be copy/pasting into my application.

Is there a public function that is intended for people to use to get load the WM_Config structure?

benpeart, 2021-05-15:
In case someone comes across this later. The simple solution I came up with was that after the standard logic to connect:

```
  // connect to wifi or enter AP mode so it can be configured
  DB_PRINT("\nStarting Kaleidoscope on " + String(ARDUINO_BOARD));
  DB_PRINTLN(ESP_ASYNC_WIFIMANAGER_VERSION);
  ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "Kaleidoscope");
  //ESPAsync_wifiManager.resetSettings();   //reset saved settings
  ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 132, 1), IPAddress(192, 168, 132, 1), IPAddress(255, 255, 255, 0));
  ESPAsync_wifiManager.autoConnect("KaleidoscopeAP");
  if (WiFi.status() == WL_CONNECTED)
  {
    DB_PRINT(F("Connected. Local IP: "));
    DB_PRINTLN(WiFi.localIP());
  }
  else
  {
    DB_PRINTLN(ESPAsync_wifiManager.getStatus(WiFi.status()));
  }

```

I added a little code to save the timezone information in the standard preferences:

```
  // this only returns a value during the initial config step (call resetSettings() to test)
  // store the string in persistant storage for later use
  String timezoneName = ESPAsync_wifiManager.getTimezoneName();
  if (timezoneName.length())
  {
    // write the timezone string into persistant memory
    DB_PRINTF("Saving timezone '%s'\r\n", timezoneName.c_str());
    const char * tz = ESPAsync_wifiManager.getTZ(timezoneName);
    Preferences preferences;
    preferences.begin("kaleidoscope", false);
    preferences.putString("tz", tz);
    preferences.end();
  }
```

Then on regular boots, I can just read the timezone from the preferences:

```
    // read the timezone from persistant memory
    Preferences preferences;
    preferences.begin("kaleidoscope", true); 
    String tz = preferences.getString("tz", "");
    preferences.end();
    if (tz.length())
    {
        configTzTime(tz.c_str(), "us.pool.ntp.org", "time.nist.gov");
        DB_PRINTF("Current timezone is %s\r\n", tz.c_str());
    }
    else
    {
        // hard code EST because I can :)
        configTime(-5 * SECS_PER_HOUR, SECS_PER_HOUR, "us.pool.ntp.org", "time.nist.gov");
        DB_PRINTLN(F("Current Timezone is not set. Enter Config Portal to set."));
    }

```

khoih-prog, 2021-05-15:
Hi Ben,

I'm glad that you've found the way out, even not the good solution.

There is much better way to do it, but I **intentionally** delay responding, so that you have to spend time to understand more how the library process is working.

It's better to start with the more complex examples, such as [Async_AutoConnect](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_AutoConnect), then trimming down than to start with minimal example, because you don't have the necessary functions to support th complex case (LittleFS, WM_Config, etc.).

Anyway, it's still better than nothing is working.

Good Luck,