khoih-prog/ESPAsync_WiFiManager #57: Auto(re)connect by using multiwifi problem

seksity, 2021-06-16:
### Describe the bug

I using multiwifi to auto(re)connect wifi router as link below:-
https://github.com/khoih-prog/ESPAsync_WiFiManager#16-using-multiwifi-autoreconnect-feature

The problem after restart wifi router, then ESP32 with ESPAsync_WiFiManager library is try to reconnect to wifi router correctly, BUT when WiFi router ready to use, the ESP32 not connect to WiFi until push RESET button and debugging message on serial port shown as below:-

> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (0).

Then no debugging message any more 

### Steps to Reproduce

1. Run library example Async_AutoConnect.ino
2. Turn-OFF Wifi Router
3. Serial port debugging message shown below message every 500 mSec (default on example code: "Async_AutoConnect.ino" -- LINE 321
  Debugging message:  

> WiFi lost. Call connectMultiWiFi in loop
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!

4. Turn-ON WiFi router and waiting ESP32 reconnect, when the WiFi router ready then ESP32 not connect to WiFi until push RESET button and stop debugging message as below:-

> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (0).

sometime like this:-

> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (0).

 
### Expected behavior

Expected ESP32 auto(re)connect to WiFi router and debugging message status like example source code, it look like below:-

> SSID: ......, RSSI= ......
> Channel: ......., IP Addresss: .......

### Actual behavior (Debugging log)
1. ESP32 not connect to WiFi until push RESET button
2. No debugging message any more and stop like below:-

> WiFi lost. Call connectMultiWiFi in loop
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
> [E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (0).

### Information

Please ensure to specify the following:

* Platform.io version 1.57.0
* `ESP32` DevModule (ESP32-WROOM-32) with ESPAsync_WiFiManager 1.9.1
* Run library example "Async_AutoConnect.ino" then Turn-Off and Turn-On WiFi router (As in ### Step to Reproduce)
* Anything that might be relevant in your opinion, such as:
  * MacOS BigSur 11.4
  * WPA/WPA2, Home networks WiFi 2.4GHz

khoih-prog, 2021-06-16:
Hi @seksity 

I think you're doing something not correctly as I tested here and everything is working as expected.
You have to enter Config Portal and add 2 sets of WiFi Credentials for MultiWiFi to work correctly.
If you have only 1 router, you can enter the 2nd set of WiFi Credentials the same as the 1st.

The issue with 1 set of of WiFi Credentials is that the MultiWiFi will clear and won't reconnect to the lost router. 
ESP32 will autoreset to be able to reconnect again.

I suggest you to use the better example [**Async_ConfigOnDoubleReset**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDoubleReset) to have more control when to enter Config Portal.

Hereafter is the debug terminal

#### 1. Working normally with 1 set of WiFi Credentials (SSID = `HueNet1`, password) entered in Config Portal

```

Starting Async_ConfigOnDoubleReset using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.9.1
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet1 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -30
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.75 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
HH
```

#### 2. Turn OFF WiFi router `HueNet1`. ESP32 resets itself

```
WiFi lost. Call connectMultiWiFi in loop
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi not connected
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
```

#### 3. WiFi router `HueNet1` turned ON. ESP32 auto-reconnect OK to previous router`HueNet1`

```
Starting Async_ConfigOnDoubleReset using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.9.1
ESP_DoubleResetDetector v1.1.1
ESP Self-Stored: SSID = HueNet1, Pass = password
[WM] * Add SSID =  HueNet1 , PW =  password
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Add SSID =  HueNet1 , PW =  password
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  password
[WM] * Add SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] * Additional SSID =  HueNet1 , PW =  password
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -29
[WM] Channel: 2 ,IP address: 192.168.2.232
After waiting 10.95 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
Stop doubleResetDetecting
Saving config file...
Saving config file OK
HHHHHHHHHH HHHHHHHHHH HHHHHHHHHH HHHHHHHHH
```

The examples are designed for MultiWiFi use cases, if you have only 1, it's better to use [very old version](https://github.com/khoih-prog/ESPAsync_WiFiManager#releases-1011) or write the code not to use MultiWiFi, just reconnect to the only WiFi SSID you have.

closed 2021-06-16 by khoih-prog

seksity, 2021-06-16:
@khoih-prog Thanks you. Latest I test with example "Async_ConfigOnDRD_ESP32_minimal.ino" to recap the minimal source code to works with 1 router, but I not sure how to add more code to support auto reconnect router?