khoih-prog/ESPAsync_WiFiManager #58: Doesn't save network credentials

Philipp66904, 2021-07-13:
### Still having this problem
The "Async_ConfigOnDoubleReset" example works fine, but the "Async_AutoConnect_ESP8266_minimal.ino" example does not. I switched to another ESP8266 and the problem is still there (so not a hardware problem).
I use this library because I want an easy and convenient way to connect to my network rather than 979 lines of code for various things that I don't need at all. So it would be nice if you could test it and maybe change the code accordingly if there is a problem with it and not on my part. Thank you.

### Describe the bug

I used following demo-programm after noticing it in my own code: "Async_AutoConnect_ESP8266_minimal.ino"
I use an ESP8266 with core v3.0.0 

### Steps to Reproduce

I upload the code to my device and it starts the config portal with AP as exepected. I enter the correct network credentials and the ESP connects to my Network without a problem.
But after restarting the ESP, it seems that it didn't save my credentials at all and I have to dial them in again. This happens over and over again without changing.

### Expected behavior

After the first time saving the credentials it should automatically connect to the network.

### Actual behavior

It asks me about my network again and says that there are no networks saved.

### Information

Please ensure to specify the following:

* Arduino IDE version 1.8.15
* `ESP8266` core v3.0.0

khoih-prog, 2021-07-13:
What can you expect from the `minimal` version of examples?

**It's not designed to save the Credentials**, just to show how to shorten the code, certainly with loosing all the necessary features.

closed 2021-07-13 by khoih-prog

bjovanovic, 2022-01-09:
Well, there was a behavior change, as the exact same code **used** to save WiFi credentials. The change occurred with ESP Core 3.0.0. The same minimal sketch saves wifi u/p on esp8266 core v2.7.4.

As far as I could test, this did not change across Async WifiManager versions, but was somehow impacted by esp8266 core update.