khoih-prog/ESPAsync_WiFiManager #59: ESPAsync_WiFiManager and ArduinoIoTCloud libraries are incompatible.

kirk-clendinning, 2021-07-28:
### Describe the bug

When trying to use an access point provided by ESPAsync_WiFiManager and the ArduinoIoTCloud functionality in the same Sketch, simply adding #include <ArduinoIoTCloud.h> to an ESPAsync_WiFiManager example project causes the access point web server to hang. This issue has been filed against ArduinoIoTCloud as well. 

### Steps to Reproduce

Add #include <ArduinoIoTCloud.h> to an ESPAsync_WiFiManager example. When the AP is accessed, the web server hangs.

### Expected behavior

I expected to be able to use ESPAsync_WiFiManager to maintain a wifi connection, and when the connection was active, use ArduinoIoTCloud to control an ESP8266 node.

### Actual behavior

When the Captive Portal is accessed by a web browser, the web server hangs and a blank page results.

### Debug and AT-command log (if applicable)

It appears that the page is rendered correctly, but is the web server hangs when it is served.

### Screenshots

It's a blank page.

### Information

Take any ESPAsync_WifiManager example and put the header #include <ArduinoIoTCloud.h> into the .ino and the captive web page will hang. 

Please ensure to specify the following:

* Arduino IDE version  1.8.15
* ESP8266 D1 mini
* Contextual information (e.g. what you were trying to achieve): I'd like to have the WiFi configuration provided by ESPAsync_WifiManager keep the ESP8266 connected so that I can use the Arduino IoT Cloud to control the device
* Simplest possible steps to reproduce: Have the ArduinoIoTCloud library available and add #include <ArduinoIoTCloud.h>  to any example project.

  

Steps to reproduce:
1. Set up a ESPAsync_WifiManager manager project such as Async_AutoConnectWithFeedbackLED, have the ArduinoIoTCloud library available and add #include <ArduinoIoTCloud.h> to the project.
2. It will compile.
3. When you try to connect to the Captured Portal, the browser will connect, but the page will be blank.

kirk-clendinning, 2021-07-29:
If NTP config is enabled in the ESPAsync_WifiManager, the captured portal web page does not render. So this comment is also advisable when using ArduinoloTCloud with ESPAsync_WiFiManager.

// Use false to disable NTP config. Advisable when using Cellphone, Tablet to access Config Portal.
// See Issue 23: On Android phone ConfigPortal is unresponsive (khoih-prog/ESP_WiFiManager#23)
#define USE_ESP_WIFIMANAGER_NTP false

khoih-prog, 2021-08-03:
Hi @kirk-clendinning 

Thanks for your interest in the library.

You have to understand that [**ArduinoIoTCloud** library](https://github.com/arduino-libraries/ArduinoIoTCloud) is a complete ecosystems for many boards, besides ESP8266, to manage the connection to `ArduinoIoTCloud`, including local SSID / PWD, and many more complex features

Nobody can expect this library, managing **Async** WebServer and connection, to co-exist with the complex synchronous `ArduinoIoTCloud library`. I don't think even the synchronous [**ESP_WiFiManager** library](https://github.com/khoih-prog/ESP_WiFiManager) can do this, without spending lots of time to study / debug and have many modifications to make them working together.

Anyway, this is **redundant** and I won't spend time on this, and don't expect anybody will do so.

Good Luck,

closed 2021-08-03 by khoih-prog

kirk-clendinning, 2021-08-03:
In the end, it appears that the conflict had to do with the NTP implementations. By disabling that in both libraries, the combination worked correctly.

Thanks for responding.

-kirk-

khoih-prog, 2021-08-03:
I'm amazed that you figure out the solution.

Thanks for your information and research, which will help many other users.

Best Regards,

kirk-clendinning, 2021-08-03:
Sadly, I did not have time to go into depth. Since the ESP8266 has no RTC, it may be that both of the libraries were trying to replace that in some way with NTP, and so there was some sort of conflict. I find the Arduino development environment a bit tedious since one cannot step for debugging. It is sometimes difficult to find issues in asynchronous libraries.

Thank you for your contributions to the community.

-kirk-