khoih-prog/ESPAsync_WiFiManager #61: Library and Opperational issues with Async_ConfigStartup sketch

vangalvin, 2021-08-18:
### Describe the bug

The first issue appeard when I was instaling the library using the Arduino Library Manager.
It seems that when I requested to also install the requred library's I get an error
no protocol: error.

However the library did install and I checked out the examples, Some work however my interest is in the Async_ConfigOnStartup.

Using the example I complied and uploded it to and it seemed to be working, Connected to the config interface and set the relevent details to connect to my router. on saving the details I started to get 

assertion "don't call tcp_abort/tcp_abandon for listen-pcbs" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 480, function: tcp_abandon

ELF file SHA256: 0000000000000000

Backtrace: 0x40088764:0x3ffb4750 0x400889e1:0x3ffb4770 0x4010013f:0x3ffb4790 0x40125925:0x3ffb47c0 0x40125a05:0x3ffb47f0 0x400e14aa:0x3ffb4810 0x40121938:0x3ffb4830 0x400899f2:0x3ffb4860

Rebooting...

### Steps to Reproduce

Using the Async_ConfigStartup sketch.ino example sketch
Have set it to use DHCP however even with a fixed IP the same issues happen 

"ESPAsync_WiFiManager v1.9.2"

Flashed the LoLin ESP32.
Connected to the ESSID on the ESP32 and set both the SSID and SSID1 with their respective passwords.

Save causes the above error.

If I however just fill in the details for SSID1 it will attempt to start but fails to connec to the network and obtain an IP.

Arduino Version 1.8.15
LoLin ESP32
ESP32 1.0.6 
ESPAsync_WiFiManager v1.9.2

khoih-prog, 2021-08-18:
Sorry that happens **only to you.** 
You have to check your system, installation, follow all the instructions and prerequisites, etc.

Nobody can solve the problem for you until you prove that this is the bug caused by this library. 
I'll close the issue now until you show the bug.

Good Luck,

closed 2021-08-18 by khoih-prog

vangalvin, 2021-08-18:
Instructions were pretty clear about how to install the libraray in to the Arduino IDE. I have even set up a new "Clean" dev machine and get the same error when trying to install the library using the library manager in the Arduino IDE.

"The best and easiest way is to use Arduino Library Manager. Search for ESPAsync_WiFiManager, then select / install the latest version. You can also use this link arduino-library-badge for more detailed instructions."

[image]

Results in the collowing error on two systems that have never had any errors and a fresh install of the Arduino IDE etc.

And installing ESPAsync_WiFiManager is the only library that gives this error, other libraries do not seem to have any issues at all. 

no protocol: 
java.lang.RuntimeException: java.net.MalformedURLException: no protocol: 
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:250)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol: 
	at java.net.URL.<init>(URL.java:593)
	at java.net.URL.<init>(URL.java:490)
	at java.net.URL.<init>(URL.java:439)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:64)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:155)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:240)
	... 1 more

khoih-prog, 2021-08-18:
Again it seems the problem is in your system (machine, virus, Java version, corrupted files / HDD, bad memory,  etc.) and I'm even patient and tried to delete then reinstalled the library via IDE. **The results are OK.**

If the library behaves the ways you described, I'm sure **nobody could use it successfully** and I have to delete it immediately.

[image]

If you like, post on Arduino forum to ask for help from some people who could experience the similar issue and have fixed.

I don't have any idea why because this seems to be **system-dependent** and I don't have a very clear crystal ball to guess.

You can also download the zip library then expand / copy into the libraries' directory.

Good Luck,