khoih-prog/ESPAsync_WiFiManager #62: Thanks!

ddTech, 2021-09-06:
Good Morning, 

I'm just stopping by to say "thank You" for that wonderful piece of software. 

I spent a good amount of yesterday working myself through the examples and bits of code to get a better understanding of what's going on and how I can use which parts for my scenario(s). I have worked with Tzapu's WifiManager before, so I kind of know what to expect and how to use the libray.
Due to the complexity and variability of what at first glance seems to be _just_ a  helpful assistant, there is a bit of a learning curve, but the sheer amount of examples seems to cover almost every possible scenario. 

Also the _Read.me_ truely is one! I must admit I  did not realize that until I already had gone through most of the examples - which are documented well enough, so that I did not expect an additional documentation. 
Most _read.me_ s here on github just cover installation and changelog or are not worth reading for other reasons. 

You might do Yourself a favour (in terms of support and trouble requests) and reorder the "Table of Contents" so that the "HOW TO"s and "Documentation" stay upfront and the "change log" moves towards the end, so that a new "customer" immediatlely sees "ah, there's more to read". 
You don't read change logs on first contact and the list is already a long one that probably will keep growing, thus moving the documentation parts even further down.

'Guess I'm ready to code now.....

Have a great day

Frank

khoih-prog, 2021-09-06:
Hi @ddTech 

Thanks for your encouraging words, and you as well as I know those are just some of many shortcomings in the library.

I wish I had the all the time in the world to make everything good enough, for more than **90** libraries I've created and am managing now. There is so much a single guy can do ;=)

> Most read.me s here on github just cover installation and changelog or are not worth reading for other reasons

This is so true, and you have to understand that for library writers, the writings of correct README.md is considered as a torture, as we are not `professional docs writers` and also have `tunnel vision` about what is necessary / good for other users.

I'd appreciate if you can help in modifying the `README.md`, certainly whenever you have time, then make a PR to contribute back. You seem to have very clear idea how to re-organize it, and I totally agree with it.

Possibly we also can move the **Changelog** section to `Changlog.md` file. A separate `wiki` for functions/classes is also good idea.

khoih-prog, 2021-09-06:
Hi,

I just tried to test-split [changelog.md](https://github.com/khoih-prog/WiFiWebServer/blob/master/changelog.md) from [README.md](https://github.com/khoih-prog/WiFiWebServer/blob/master/README.md) for [WiFiWebServer Library](https://github.com/khoih-prog/WiFiWebServer), and can see a better / clearer look for README.

ddTech, 2021-09-07:
Good morning, 

> I just tried to test-split [changelog.md](https://github.com/khoih-prog/WiFiWebServer/blob/master/changelog.md) from [README.md](https://github.com/khoih-prog/WiFiWebServer/blob/master/README.md) [...]

yes, that looks good. I think it helps to have the documenation separate.

Meanwhile I got my current project running with the Async WiFi Manager, awsome! 
I love Your Pin-Assignment-Constants-list, as I had to look up the pins up again and again. I exported it to a separate h-file. :-)

I also moved some parts to separate functions() which gives a cleaner structure. So all the FS-stuff has now moved to a function mountFS(),  I'm not done yet with everything, as I not fully understand the meaning / reason of every line (yet). 
_Async_ConfigPortalParamsOnSwitch.ino_ for instance seems to have basicly the same code in the setup() as in the loop() and within that the _USE_ESP_WIFIMANAGER_NTP_ section with the configTzTime(), which is also used from within check_status().
not sure yet if they are truely the same and interchangable, but I'll see if some of that could be refactored for a cleaner structure and better readability. 

I'm also weighing the pros and cons of having esp8266 and esp32 in one sketch. It allows to easily use the code on both processors, but makes it much much harder to read and maintain, especially for an outsider new to the matter.

Stuff like this is painful to read, even in platformio, which I am using.
```
  // Format FileFS if not yet
#ifdef ESP32
  if (!FileFS.begin(true))
#else
  if (!FileFS.begin())
#endif
  {
#ifdef ESP8266
    FileFS.format();
#endif
```

Took me a moment to find the matching bracket. You are using single line if-else-endif statements without brackets, which in this case makes it much harder for the eye to catch where certain bits of code belong to.

I tried to modify it like so: 
```
#ifdef ESP32
  if (!FileFS.begin(true)){
#else
  if (!FileFS.begin()){
    FileFS.format();
#endif
```

but then vscode is unable to see the matching bracket. So it has to stay like it was.
Switching between #ifdef ESP32 and #ifdef ESP8266 does not make it better, but is partly inevitable as well.

I don't have _the_ solution to that, Maybe in cases where code constantly changes betweeen includes and excludes it would be better to have separate blocks with partly duplicate code but distinctive task. Something like this
```
#if ESP32
  mountFS_ESP32();
#else
  mount FS_ESP8266();
#endif
```

In the end the compiler does not care and the code is running fine :-). 

Ah, I found some missing function prototypes: 
```
loadConfigData();
saveConfigData();
check_status();
check_WiFi();
toggleLED();
```

I think, the Arduino IDE does not care, but PIO complains. 

> This is so true, and you have to understand that for library writers, the writings of correct README.md is considered as a torture, as we are not `professional technical writers` and also have `tunnel vision` about what is necessary / good for other users.
> I'd appreciate if you can help in modifying the `README.md`, certainly whenever you have time, then make a PR to contribute back. You seem to have very clear idea how to re-organize it, and I totally agree with it.

Exactely. For my main product, a scheduling application (nothing to do with microcontrollers) I have written hundreds of pages of documentation. Every sentence painfully "created", lots fo screenshots. ... By the time I am finished with a topic another one has to be modified, because development has mooved on. So I_m constantly _behind_  .... and then people don't even look at it. As they are either are (or hopefully become) paying cusomers, You don't simply tell them "RTFM!" but take their hand and say "come, I'll show You".

I'd like to assist and contribute, but am, just like You, trying to joggle all my own projects and keep them up in the air while the ones I'd really like to work on still lay on the floor waiting. 
Additionally, I am a long term all time developer, but never got a hand to github, as there was no necessity to use it (yet). So I'm not sure if I could help if I wanted to. On the other hand, one has to start somehow :-). Anyway, I'm glad You easily managed to modify the read.me. 

Have a great day!

Regards

Frank

khoih-prog, 2021-09-09:
Done splitting changelog from README. Better now !!!  ;=))

Thanks,

closed 2021-09-09 by khoih-prog