khoih-prog/ESPAsync_WiFiManager #65: Use NTP server given by DHCP server option 42

fow0ryl, 2021-09-27:
I would like to use the ntp server address given by a local dhcp server.
If the DHCP server is configured with option 42 all devices in the LAN/VLAN can use that address.
No problems with changing addresses of local ntp-servers or blocking firewalls anymore ...

Henning

khoih-prog, 2021-09-27:
Have a look at this [**Tips on using NTP-Services** #63](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/63)

closed 2021-09-27 by khoih-prog

khoih-prog, 2021-09-27:
If you have assigned your local NTP server a `hostname`, such as `My_NTP_Server`, then you can use that name instead of "time.nist.gov" or "0.pool.ntp.org", and if your local router DHCP is working OK, you can just use that name and **hopefully** the local NTP address will be resolved correctly as this really depends on the router / GW you're using.

For ESP, hopefully you can use **mDNS** to resolve NTP-server IP address from local `hostname` via `Bonjour` service. Check [**MDNS_Generic** Library](https://github.com/khoih-prog/MDNS_Generic)

Another way to use is **DDNS + UPnP.** Check [**UPnP_Generic** Library](https://github.com/khoih-prog/UPnP_Generic). TO use this option, you must have DDNS address, such as domain name acquired from duckdns, ddns, NoIP, etc.

Certainly, this is slower and not as independent as the other option of using the Dynamic Parameters after you've known the IP address of your local NTP.