khoih-prog/ESPAsync_WiFiManager #70: library.properties "depends" statement inconsistent with intention

lbussy, 2021-11-01:
### Describe the bug

The `library.properties` file is here:

```
name=ESPAsync_WiFiManager
version=1.9.4
author=Khoi Hoang
maintainer=Khoi Hoang <khoih.prog@gmail.com>
license=MIT
sentence=ESP32 (including ESP32-S2 and ESP32-C3), ESP8266 WiFi Connection Manager using AsyncWebServer, with enhanced GUI and fallback Web ConfigPortal.
paragraph=This Library is used for configuring ESP32 (including ESP32-S2 and ESP32-C3), ESP8266 modules WiFi Credentials at runtime. You can also specify static DNS servers, personalized HostName, fixed or random AP channel. Now with MultiWiFi auto(Re)connect, configurable CORS Header and auto-Timezone features.
category=Communication
url=https://github.com/khoih-prog/ESPAsync_WiFiManager
architectures=esp8266,esp32
depends=ESP Async WebServer,ESP AsyncTCP,AsyncTCP,ESP_DoubleResetDetector
includes=ESPAsync_WiFiManager.h
```

The `depends=` statement includes the "standard" `ESP Async WebServer`, `ESP AsyncTCP`, and `AsyncTCP` which causes PlatformIO to automatically download these libraries.  When it does this, `ESP Async WebServer` is the upstream tzapu version, and conflicts with your own version:

```
da---l         11/1/2021   8:42 AM                ESP Async WebServer
da---l         11/1/2021   8:42 AM                ESP Async WebServer@src-e4ae869850c1f195efe6c56e35c37656
```

It also forces downloading both of the AsyncTCP versions, only one or the other is needed.  

I *believe* this is a carry-over from the upstreal `library.properties` file.

### Steps to Reproduce

Use the library within PlatformIO.

### Expected behavior

I expect since the documentation instructs one to include those libraries, that the core ESPAsync_WiFiManager not enforce these dependencies and cause potential conflicts.

khoih-prog, 2021-11-02:
Thanks for your interest in the library.

Could you please check PIO:

1. [Library Management](https://docs.platformio.org/en/latest/librarymanager/index.html)

> Library Dependency Finder can be configured from “platformio.ini”

2. [Compatibility Mode](https://docs.platformio.org/en/latest/librarymanager/ldf.html#compatibility-mode)

> Compatibility mode allows one to control strictness of Library Dependency Finder. If library contains one of manifest file (**library.json**, library.properties, module.json), then LDF check compatibility of this library with real build environment. 

and understand the way PIO is managing libraries. `library.json` has higher priority than `library.properties` if both exist.

**PIO requires much more knowledge and skills** to understand and manually modify and change the settings. I don't have time to spend on your PIO issue and it's better that you ask for help in [PIO forum](https://community.platformio.org/).

Good Luck,

closed 2021-11-02 by khoih-prog

lbussy, 2021-11-02:
So you forked someone's lib, you carried over their settings, and now you're telling me it's my problem I used your fork of this lib?

Ever work at Apple?

I know how to ignore the files you can;t seem to be able to configure correctly.  I incorrectly thought you'd care.