khoih-prog/ESPAsync_WiFiManager #71: Click focus js bug on wifi page

idkq, 2021-11-07:
### Describe the bug

Weird behavior with focus. When I click on the name of the network link it populates SSID and SSID1 fields and move the focus to SSID1 password. When I start typing the password, the first char populates the first password box and then the focus moves to the second password box.

### Steps to Reproduce

On page wifi:

1. Click on the link of the network (in my case first link)
2. Type password.

You should see the first char in password box 1 and the rest on password box 2

[image]

Also why does it populate 2 fields? Why do we have 4 boxes for user/pwd? I assume if user wants to add additional networks if one fails? How to remove SSID/pwd boxes?

idkq, 2021-11-09:
Tested on safari and chrome

khoih-prog, 2021-11-10:
Hi,

Thanks for your interests in the library.

It seems that your issue is strange and unique, system dependent as I can't duplicate here, using Chrome and Ubuntu Linux.

[image]

---

> Also why does it populate 2 fields? Why do we have 4 boxes for user/pwd? I assume if user wants to add additional networks if one fails? How to remove SSID/pwd boxes?

I'm frustrated when the good feature, requested and useful to many people, is not fully understood and considered redundant. Anyway, nothing can fit all.

I suggest you to read [**Major Releases v1.1.1**](https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/changelog.md#major-releases-v111) and use pre-v1.1.1 if requiring only 1 SSID. Or you can even ignore the 2nd SSID by leaving it blank.

There is no easy way to remove the MultiWiFi feature in releases v1.1.1+ and it's not recommended.

closed 2021-11-10 by khoih-prog