khoih-prog/ESPAsync_WiFiManager #72: WiFiManager "on demand" (for configuration only) example

ClemensGruber, 2021-11-15:
I'm searching for example code that uses WiFi "on demand" for configuration only. Data telemetry is done by GSM (and only optional by Wifi). So the scenario is 

- Normally Wifi sould be not activated to save power, I have a solar powered device that makes nothing 99 % of the time, that means deep sleep. 
- But even if the UI and Wifi does not run, I need accesss to the variables modified and configured via the WifiManager.
- In case configuration mode / on demand is activated we need a timeout, in case someone forget to disconnect from the the UI the device should stopp the UI and WiFi after a time and go in deep sleep, otherwise the device will run out of battery.

Is there any exampel code for this scenario? I'm a bit confused about the many Async_ConfigOn... examples ...

khoih-prog, 2021-11-15:
Hi,

Thanks for your interest and trust in the library.
As you know, no library and example is designed to be able to apply in all complex use cases in the world, especially yours.

I'm sorry I won't spend time on this not-issue, unless a library's bug is proven.

I suggest you ask for advice of many experts in [Arduino Forum](https://forum.arduino.cc/) and/or [ESP32](https://www.esp32.com/viewforum.php?f=23) / [ESP8266](https://www.esp8266.com/viewforum.php?f=25) forums

Good Luck,

closed 2021-11-15 by khoih-prog

khoih-prog, 2021-11-15:
BTW, have a look at these examples to see if you can use anything

1. [**SmartFarm_DeepSleep**](https://github.com/khoih-prog/SmallProjects/tree/master/SmartFarm_DeepSleep)
2. [**Async_ConfigOnSwitchFS**](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnSwitchFS) where you can replace this following line [**Async_ConfigOnSwitchFS.ino**#L1201](https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/examples/Async_ConfigOnSwitchFS/Async_ConfigOnSwitchFS.ino#L1201) with any condition you'd like to activate Config Portal.

```
if ((digitalRead(TRIGGER_PIN) == LOW) || (digitalRead(TRIGGER_PIN2) == LOW))
```

Certainly, your use case requires more careful and complex design and implementation.