khoih-prog/ESPAsync_WiFiManager #74: Not working with ESP32 and Core 2.0.1 (or 2.0.0+)

Frtrillo, 2021-11-26:
### Describe the bug

When you compile any example on ESP32 and core 2.0.1 you get  LittleFS errors.

### Steps to Reproduce

Compile any example on an ESP32 with core 2.0.1
### Expected behavior

Works and compile. (its works with esp8266 though)
### Actual behavior

Throws several LittleFS errors. 

### Debug and AT-command log (if applicable)

Also tried with full-clean computer and got the same issue.

khoih-prog, 2021-11-26:
Hi @Frtrillo 

Thanks for your interest in the library.

I've just test compiled and there is no issue at all. I'm using ESP32 core v2.0.1

[image]

[image]

I think you have some issue with your installation and only you can find out and fix the issue.

Good Luck,

khoih-prog, 2021-11-26:
Hi @Frtrillo 

I think I found out the issue.

You need to install [**Lorol's LITTLEFS** library](https://github.com/lorol/LITTLEFS) and don't change the example code.

I'll update several libraries in the chain (including [**ESP_DoubleResetDetector**](https://github.com/khoih-prog/ESP_DoubleResetDetector), etc.) to use the ESP32 core's built-in LittleFS, instead of [**Lorol's LITTLEFS** library](https://github.com/lorol/LITTLEFS), which has been absorbed into the core since v1.0.6

I'm opening the issue now. Please wait.

khoih-prog, 2021-11-26:
Hi @Frtrillo 

The new [**ESPAsync_WiFiManager releases v1.9.5**](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.9.5) has just been published to fix the issue.

Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/ESPAsync_WiFiManager#contributions-and-thanks)

Please try and verify the issue is fixed.

---

### Releases v1.9.5

1. Auto detect ESP32 core and use either **built-in LittleFS** or [**LITTLEFS**](https://github.com/lorol/LITTLEFS) library
2. Update `library.json` to use new `headers` for PIO

closed 2021-11-26 by khoih-prog

ntableman, 2022-03-25:
I am seeing this happen on my setup, I have the latest versions of the libraries listed in the prerequisites and ESP32-S2 (Adafruit feather). I can compile and run the various FS examples, so I know my compiler/env and the board is working. I try to trace through the if's to switch between the various boards and cores, but in the end I get to : 

13:00:38.765 -> Starting Async_AutoConnectAP using LittleFS on ADAFRUIT_FEATHER_ESP32S2
13:00:38.765 -> ESPAsync_WiFiManager v1.12.2
13:00:38.765 -> SPIFFS/LittleFS failed! Already tried formatting.
13:00:38.864 -> LittleFS failed!. Please use SPIFFS or EEPROM. Stay forever

ESP v 2.0.2
ESP_DoubleResetDetector v 1.3.1

khoih-prog, 2022-03-26:
I suggest you create a new issue if there is one or just post on [**Discussions**](https://github.com/khoih-prog/ESPAsync_WiFiManager/discussions) if there is something you can't do and not be sure that's a bug of the library.

Please don't **hijack** another closed issue.

Anyway, I just retested and verify everything is so far **OK** as displayed in the following terminal outputs. Check if your **boards, ESP32 core or settings** is correct. I'm not going ti waste my time here, unless you prove there is a bug of the library and open a new issue.

Good Luck,

---

### Terminal Ouput

#### Start normally with LittleFS (already formatted)

```
Starting Async_AutoConnectAP using LittleFS on ESP32S2_DEV
ESPAsync_WiFiManager v1.12.2
[WM] RFC925 Hostname = AutoConnectAP
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 0.0.0.0 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
[WM] Current TZ_Name = America/New_York , TZ =  EST5EDT,M3.2.0,M11.1.0
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet1 , PW =  12345678
[WM] * Additional SSID =  HueNet2 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -22
[WM] Channel: 4 ,IP address: 192.168.2.155
After waiting 9.38 secs more in setup(), connection result is connected. Local IP: 192.168.2.155
[WM] freeing allocated params!

```

### Restart with SPIFFS => Re-format SPIFFS

```
Starting Async_AutoConnectAP using SPIFFS on ESP32S2_DEV
ESPAsync_WiFiManager v1.12.2
E (592) SPIFFS: mount failed, -10025
[WM] RFC925 Hostname = AutoConnectAP
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 0.0.0.0 , gatewayIP = 0.0.0.0
[WM] netMask = 0.0.0.0
[WM] dns1IP = 0.0.0.0 , dns2IP = 0.0.0.0
Got stored Credentials. Timeout 120s for Config Portal
Current Timezone is not set. Enter Config Portal to set.
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -22
[WM] Channel: 4 ,IP address: 192.168.2.155
After waiting 6.96 secs more in setup(), connection result is connected. Local IP: 192.168.2.155
[WM] freeing allocated params!
```

#### Restart with LittleFS => Re-format LittleFS

```
Starting Async_AutoConnectAP using LittleFS on ESP32S2_DEV
ESPAsync_WiFiManager v1.12.2
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
E (601) esp_littlefs: mount failed,  (-84)
E (605) esp_littlefs: Failed to initialize LittleFS
[WM] RFC925 Hostname = AutoConnectAP
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] failed
Open Config Portal without Timeout: No stored Credentials.
We haven't got any access point credentials, so get them now
Starting configuration portal @ 192.168.4.1, SSID = ESP_8A1DF7C_AutoConnectAP, PWD = MyESP_8A1DF7C
[WM] 
Configuring AP SSID = ESP_8A1DF7C_AutoConnectAP
[WM] AP PWD = MyESP_8A1DF7C
[WM] AP Channel = 5
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] startConfigPortal : Enter loop
[WM] Can't use Custom STA IP/GW/Subnet
[WM] Connected after waiting (s) : 1.40
[WM] Local ip = 192.168.2.155
[WM] Timed out connection result: WL_CONNECTED
WiFi connected...yeey :)
[WM] Current Timezone Name is not set. Enter Config Portal to set.
[WM] getSTAStaticIPConfig
[WM] SaveWiFiCfgFile 
[WM] stationIP = 0.0.0.0 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.155
[WM] freeing allocated params!
```