khoih-prog/ESPAsync_WiFiManager #83: Possibility to configure credentials without connecting

perotom, 2022-01-17:
### Is your feature request related to a problem? Please describe.

I want to enter credentials to a WiFi which is not currently present. If I do so `startConfigPortal` never returns and in my logs I see:
`18:55:38.107 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND`
Is there any way to just skip the connection test?

khoih-prog, 2022-01-18:
> Is there any way to just skip the connection test?

It's not `connection test`, just a message after it failed to connect to the not-existing AP.

What's else can you expect?

---

```
Starting Async_ConfigOnDoubleReset_TZ using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.10.0
ESP_DoubleResetDetector v1.2.1
ESP Self-Stored: SSID = HueNet1, Pass = jenniqqs
[WM] * Add SSID =  HueNet1 , PW =  jenniqqs
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D01234
doubleResetDetected
Saving config file...
Saving config file OK
Open Config Portal without Timeout: Double Reset Detected
Starting configuration portal @ 192.168.4.1, SSID = ESP_8ABF498, PWD = MyESP_8ABF498
[WM] Connecting to new AP               <==== Powered off AP == not present AP
[WM] Failed to connect
[WM] Connecting to new AP               <==== Powered off AP == not present AP
[WM] Failed to connect
[WM] Connecting to new AP               <==== Powered off AP == not present AP
...

WiFi connected...yeey :)               <==== Powered restored AP
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet3 , PW =  12345678
[WM] SaveWiFiCfgFile 
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
HH
```

closed 2022-01-18 by khoih-prog

perotom, 2022-01-18:
Oke thanks for clarification but startConfigPortal never returns and therefore I cannot save credentials. Why is that?
I appreciate your help!

khoih-prog, 2022-01-18:
You're on your own. Try to run, understand the working examples' before adapt them to your use case. Then ask why.

perotom, 2022-01-18:
I tried the minimal example. Doesn't work.

```
#include <Arduino.h>

#define USE_AVAILABLE_PAGES         false
#define USE_ESP_WIFIMANAGER_NTP     false

#include <ESPAsync_WiFiManager.h>
#include <ESPAsync_WiFiManager-Impl.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");

  AsyncWebServer webServer(80);
  DNSServer dnsServer;
  ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "ESP32");
  ESPAsync_wifiManager.setAPStaticIPConfig(IPAddress(10,0,0,1), IPAddress(10,0,0,1), IPAddress(255,255,255,0));
  ESPAsync_wifiManager.setConfigPortalChannel(0);
  if (!ESPAsync_wifiManager.startConfigPortal("ESP32")) {
    Serial.println(F("Not connected to WiFi"));
  } else {
    Serial.print(F("Local IP: ")); Serial.println(WiFi.localIP());
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

I manually enter a SSID and password and press save. It says "Try connecting ESP to the Thomas iPhone/ network. Wait around 10 seconds then check if it's OK.".

ESP output is:

```
17:43:37.445 > Starting
17:43:37.569 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
17:43:37.574 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:43:37.583 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
17:43:40.299 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
17:43:46.227 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
17:43:47.339 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
17:44:13.225 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:13.230 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:15.401 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:15.407 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:15.494 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:17.522 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:19.858 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:19.864 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:19.954 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:22.044 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:24.382 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:24.387 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:24.476 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:26.553 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:28.889 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:28.894 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:28.984 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:31.011 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:33.212 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:33.217 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:33.308 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:35.339 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:37.526 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:37.532 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:37.622 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:39.652 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:41.989 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:41.995 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:42.085 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:42.834 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 17 - AP_STADISCONNECTED
17:44:44.177 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:46.314 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:46.319 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:46.410 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:46.417 > [E][WiFiUdp.cpp:183] endPacket(): could not send data: 5
17:44:48.488 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:51.024 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:51.030 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:51.120 > [E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 5
17:44:51.120 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:53.144 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:55.545 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:55.551 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:55.641 > [E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 5
17:44:55.641 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:44:57.668 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:44:59.854 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
17:44:59.859 > [W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:44:59.950 > [E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 5
17:44:59.950 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
17:45:01.977 > [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
```

The function `startConfigPortal` never returns so my sketch is forever locked.

khoih-prog, 2022-01-23:
Never use `minimal` example as it was written to show how short you can use the library, certainly **without any good feature**.

Try this [**Async_ConfigOnDoubleReset** example](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ConfigOnDoubleReset) and see how it's working for you before wasting more time here.

I'll delete all minimal examples in next release to avoid confusing people and wasting time.

Good Luck,