khoih-prog/ESPAsync_WiFiManager #85: when FIX IP is set in WiFiManager winsock client can not connect to server

microicRI, 2022-02-15:
### Describe the bug

when wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn); //is active in program
if (client.connect(server_na_koji_se_spaja, 27015)){ // give error
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net

### Expected behavior

to work with fix IP as well as with DHCP value IP

* Arduino IDE version (e.g. 1.8.18) 
* `ESP32` core 2.0.2

### Example

```
Arduino IDE version: 1.8.18
ESP32 Core Version `2.0.2`
OS: Windows10
`// Import required libraries
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "SPIFFS.h"
#include <SPI.h>
#include <Wire.h>
#include <ESPmDNS.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

#include <AsyncElegantOTA.h>
//#include <WebSocketClient.h>

// Replace with your network credentials

const char* ssid = "kuci";
const char* password = "kucikuci01";

String remDev = "";
String serPsswd = "";
//String remDev = "LjudevitaMatesica-1";
//String serPsswd = "*Oknardaj1!*";
//const char *ssid = "kuci";     // replace with your wifi ssid and wpa2 key
//const char *pass = "kucikuci01";
//const char *server1 = "jadrand.ddns.net";
//char host[] = "jadrand.ddns.net";
#define LED_BUILTIN 16
#define SENSOR  2

long currentMillis = 0;
long previousMillis = 0;
int interval = 1000;
boolean ledState = LOW;
float calibrationFactor = 4.5;
volatile byte pulseCount;
byte pulse1Sec = 0;
float flowRate;
unsigned long flowMilliLitres;
unsigned int totalMilliLitres;
float flowLitres;
float totalLitres;
int pressCO2;

bool shouldSaveConfig = false; //manipulira se iz WiFi Mangera
char lokacija_uredjaja[20];
char broj_uredjaja[2];
char server_na_koji_se_spaja[20];
char password_za_server[20];
//default custom static IP
char static_ip[16] = "192.168.1.22";
[... 437 lines cut ...]
connected to choosed WiFi ;)
WiFi Connected to: kuci
WiFi Signal strength: -67
WiFi connected with IP: 192.168.1.133
SETUP: uzimanje vrijednosti iz upisanog podatka sa wifi manager stranice 
SETUP: kopirani podaci sa web portala u varijable: 
lokacija_uredjaja = LjudevitaMatesica
broj_uredjaja = 1
server_na_koji_se_spaja = jadrand.ddns.net
password_za_server = *Oknardaj1!*
local ip
192.168.1.133
192.168.1.1
255.255.255.0
DNS= VODA-1
mDNS responder started
Flow rate: 0.00L / min	Output Liquid Quantity: 0mL / 0.00L
server_na_koji_se_spaja= jadrand.ddns.net
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
Not connected to server !
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
Not connected to server !
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
`

khoih-prog, 2022-02-15:
From 

```
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
```

It seems the DNS can not resolve the hostname `jadrand.ddns.net` and you're using ESP32 core **v1.0.5** (not using ESP32 core **v2.0.2**) because

https://github.com/espressif/arduino-esp32/blob/c4954dd582f5bde478316ddf184ded6979e812b3/libraries/WiFi/src/WiFiGeneric.cpp#L1323

1. Check if you can access `jadrand.ddns.net` from your network by pinging from a terminal
2. Try to use the IP directly to verify your code is OK
3. Update to **ESP32 core v2.0.2**

Try to write a simple program to make sure you can communicate with `jadrand.ddns.net`  before moving forward.

I'm closing the issue now because this is not the issue of this library.

---

PS:

In order not to waste time, please post the **correct info** as well as the [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example).

I'll delete the future issue if not following the minimum standard.

closed 2022-02-15 by khoih-prog

khoih-prog, 2022-02-16:
I just saw your post in Arduino Forum

[**ESP32 + wifi manager + winsock client + fix ip NOT WORKING**](https://forum.arduino.cc/t/esp32-wifi-manager-winsock-client-fix-ip-not-working/959658) and you already tested with other libraries and are sure the issue is not caused by this library.

khoih-prog, 2022-02-16:
Moreover, you're not using this library at all.

```
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "SPIFFS.h"
#include <SPI.h>
#include <Wire.h>
#include <ESPmDNS.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
```

Please, next time, post the issue to the correct related library to not abuse the free time of other people.