khoih-prog/ESPAsync_WiFiManager #86: Connect to WiFi AP using WPS

mawi42, 2022-02-22:
Sure you can start the config portal and configure the WiFi connection. However since there is WPS supported by ESP libs I guess it would be nice before starting config portal to check if WPS connection is possible (e.g. optional).
Is there any plan to add this feature?

khoih-prog, 2022-02-22:
Hi @mawi42 

Thanks for your interest in the library and the enhancement request.

I have no plan now to add the `WPS` feature to the library, but if there are many more requests, I certainly will rethink.

Regards,

closed 2022-02-22 by khoih-prog

mawi42, 2022-02-23:
> Hi @mawi42 [1]
> Android [5].

Hi,

thank you for the fast feedback.

In case I implement this feature are you interested to receive it as 
pull request?

Best regards

Matthias

Links:
------
[1] https://github.com/mawi42
[2] 
https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/86#issuecomment-1048246381
[3] 
https://github.com/
[4] 
[5]

khoih-prog, 2022-02-23:
Certainly your PR will be appreciated and merged when proven OK.

mawi42, 2022-02-27:
Hi again,

so I started to look in more detail into your lib and have to say it is 
not so easy to find a good logic and place to add the WPS easily. So for 
now I will just use it as it is and will not add - I just want to let 
you know.

Thank you anyway for the nice lib

Best regards

Matthias

Links:
------
[1] https://github.com/mawi42
[2] 
https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/86#issuecomment-1048246381
[3] 
https://github.com/
[4] 
[5]

khoih-prog, 2022-02-27:
I'd appreciate it if you can post the sample **WPS** code or link so that if I have some time, I can try to include it in any of my WiFiManager libraries.

Thanks for your try and  contribution so far,

Best Regards,