khoih-prog/ESPAsync_WiFiManager #93: Build Error - no matching function for call to 'LITTLEFSImpl::open

crashmatt, 2022-06-09:
### Describe the bug
Mismatching parameters on call to LITTLEFSImpl::open in LITTLEFS.cpp causes failure to build

### Steps to Reproduce
Create empty project/folder
Create main with empty setup and loop and basic 
Use  PIO Home to search for and add ESPAsync_WiFiManager
Build

Have tested with a code samples and  empty main with basic includes.
Tested with different filesystems defined in platformio.ini
Tested with different filesystems defined main

### Expected behavior
To build

### Actual behavior
Build error:
.pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'

### Debug and AT-command log (if applicable)
Not building

### Screenshots

### Information

`
#include "FS.h"
#include <Arduino.h>
#include "WiFi.h"
#include "LittleFS.h"

void setup()
{
};

void loop()
{
};
`

Building on VSCode Ubuntu20.04

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing Hash
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsync_WiFiManager @ 1.12.2
|   |-- AsyncTCP @ 1.1.1
|   |-- ESP Async WebServer @ 1.2.3
|   |   |-- AsyncTCP @ 1.1.1
|   |   |-- FS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ESP_DoubleResetDetector @ 1.3.1
|   |   |-- LittleFS_esp32 @ 1.0.6
|   |   |   |-- FS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- LittleFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |   |-- LittleFS_esp32 @ 1.0.6
|   |   |   |   |-- FS @ 2.0.0
|   |-- LittleFS_esp32 @ 1.0.6
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS_esp32 @ 1.0.6
|   |   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32dev/lib9d8/LittleFS_esp32/LITTLEFS.cpp.o
.pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp: In member function 'virtual bool LITTLEFSImpl::exists(const char*)':
.pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp:44:28: error: no matching function for call to 'LITTLEFSImpl::open(const char*&, const char [2])'
     File f = open(path, "r");
                            ^
In file included from .pio/libdeps/esp32dev/LittleFS_esp32/src/LITTLEFS.cpp:17:
/home/matt/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.h:38:17: note: candidate: 'virtual fs::FileImplPtr VFSImpl::open(const char*, const char*, bool)'
     FileImplPtr open(const char* path, const char* mode, const bool create) override;
                 ^~~~
/home/matt/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.h:38:17: note:   candidate expects 3 arguments, 2 provided

crashmatt, 2022-06-09:
Found the error in LITTLEFS issues:
https://github.com/lorol/LITTLEFS/issues/43

khoih-prog, 2022-06-16:
It's better to use the new ESP32 core v1.0.6+, such as ESP32 core v2.0.3 in PIO.

The [lorol's **LITTLEFS**](https://github.com/lorol/LITTLEFS) library has been deprecated and I'm afraid won't be updated to fix that bug.

closed 2022-06-16 by khoih-prog

crashmatt, 2022-06-16:
Yes.  I  figured out that I should use SPIFFS.  Trying to learn too many things at the same time.
Thanks for the response.
/Matt