khoih-prog/ESPAsync_WiFiManager #95: SPIFFS failed!. Please use LittleFS or EEPROM. Stay forever

Petermarcu, 2022-06-20:
### Describe the bug

I am trying to run the example Async_ConfigOnStartup on my ESP32_DEV board and hit the "SPIFFS failed!. Please use LittleFS or EEPROM. Stay forever" error. I'm hoping you can point me in the right direction to debug further or someone might know what I'm doing wrong.

### Steps to Reproduce

Run Async_ConfigOnStartup example on ESP32_DEV  board

### Expected behavior

AP becomes visible when scanning from my browser enabled device. 

### Actual behavior

I noticed that it is never getting to the point of setting up the AP. When I open the serial monitor and hard reset the device, I see this error.

```
Starting Async_ConfigOnStartup using FFat on ESP32_DEV
ESPAsync_WiFiManager v1.12.2
SPIFFS/LittleFS failed! Already tried formatting.
SPIFFS failed!. Please use LittleFS or EEPROM. Stay forever
```

### Debug and AT-command log (if applicable)

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Information

I just installed all the latest of everything. Arduino IDE, libraries. Running on Windows.

khoih-prog, 2022-06-20:
Hi @Petermarcu 

Sorry I can't duplicate your issue. 

```
Starting Async_ConfigOnStartup using LittleFS on ESP32_DEV
ESPAsync_WiFiManager v1.12.2
[WM] RFC925 Hostname = AsyncConfigOnStartup
[WM] Set CORS Header to :  Your Access-Control-Allow-Origin
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
Opening configuration portal.
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] failed
Open Config Portal without Timeout: No stored Credentials.
Starting configuration portal @ 192.168.4.1, SSID = ESP_9ABF498, PWD = your_password
[WM] 
Configuring AP SSID = ESP_9ABF498
[WM] AP PWD = your_password
[WM] AP Channel = 3
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] startConfigPortal : Enter loop
[WM] Request redirected to captive portal
[WM] Location http:// 192.168.4.1
[WM] Request redirected to captive portal
[WM] Location http:// 192.168.4.1
[WM] Request redirected to captive portal
[WM] Location http:// 192.168.4.1
[WM] Request redirected to captive portal
[WM] Location http:// 0.0.0.0
[WM] Request redirected to captive portal
[WM] Location http:// 192.168.4.1
[WM] Can't use Custom STA IP/GW/Subnet
[WM] Connected after waiting (s) : 1.30
[WM] Local ip = 192.168.2.109
[WM] Timed out connection result: WL_CONNECTED
WiFi connected...yeey :)
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
[WM] getSTAStaticIPConfig
[WM] SaveWiFiCfgFile 
[WM] stationIP = 0.0.0.0 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.109
[WM] freeing allocated params!
H
```

You have to check, step-by-step

- your hardware
- update to ESP32 core **v2.0.3** (it seems you're using old ESP32 core)
- test littlefs / file system to be sure OK before using this library

I suggest next time you post the problem on **Discussions**, not **Issues** if it is not a bug of the library.

I'm closing the issue now as not a bug.

Good Luck,

closed 2022-06-20 by khoih-prog

Petermarcu, 2022-06-20:
Thanks. Will do that next time and will follow your suggestions.

Petermarcu, 2022-06-22:
FYI - The version of the ESP core was the issue. Glad you noticed that. I didn't realize that the package config I was using wasn't the latest. It only had versions up to 1.0.6. After updating to 2.0.3 things just worked.

khoih-prog, 2022-06-22:
Thanks for your info which will help many other users to avoid the same issue.

Regards,

soenkegissel, 2022-08-26:
How do i update esp32 core in platformio?

khoih-prog, 2022-08-26:
Hi @soenkegissel 

Check

1. [**PlatformIO OSS July Updates: PlatformIO Core 6.1, New boards & dev-kits, Arduino Core for ESP32 v2.0.4 and Teensyduino v1.57**](https://community.platformio.org/t/platformio-oss-july-updates-platformio-core-6-1-new-boards-dev-kits-arduino-core-for-esp32-v2-0-4-and-teensyduino-v1-57/29033)
2. [**Change arduino-esp32 core version**](https://community.platformio.org/t/change-arduino-esp32-core-version/24593)
3. [**PlatformIO Open Source July Updates**](https://piolabs.com/blog/news/platformio-oss-july-2022-updates.html)

There are many more answered issues about this on the PIO Forum