khoih-prog/ESPAsync_WiFiManager #97: Device reboot when webServer.begin(); is called

LiveRock, 2022-07-22:
### Describe the bug

I copied the code from 
https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/master/examples/Async_ESP32_FSWebServer/Async_ESP32_FSWebServer.ino

and integrate it in my own program. When the line webServer.begin(); is called, the device reboot Everytime.

### Steps to Reproduce

As above

### Expected behavior

A web server will be started and behave as described in the readme in the link

### Actual behavior

Device reboot

### Debug and AT-command log (if applicable)

A clear and concise description of what you expected to happen.

### Screenshots

[WM] freeing allocated params!
CORRUPT HEAP: Bad head at 0x3ffb3a70. Expected 0xabba1234 got 0x401f81f4
abort() was called at PC 0x4008a025 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f4a4:0x3ffb3490 0x4008f71d:0x3ffb34b0 0x4008a025:0x3ffb34d0 0x4008a151:0x3ffb3500 0x4016da9f:0x3ffb3520 0x40167051:0x3ffb37e0 0x40166f41:0x3ffb3830 0x40093f1d:0x3ffb3860 0x40082a56:0x3ffb3880 0x40089f1d:0x3ffb38a0 0x4000bec7:0x3ffb38c0 0x401ce5c5:0x3ffb38e0 0x40136241:0x3ffb3900 0x401f81ff:0x3ffb3920 0x4013deab:0x3ffb3940 0x4013e222:0x3ffb3960 0x4013e2be:0x3ffb39a0 0x400dcdda:0x3ffb39c0 0x400dda19:0x3ffb3c60 0x400de2bd:0x3ffb3cd0 0x4014baf6:0x3ffb3fb0 0x40090722:0x3ffb3fd0

Rebooting...

### Information

Please ensure to specify the following:

* Platform.io version Code 6.1.3 Home 3.4.2
* ESP32 TTGO T8 - 'Wrover'
* Contextual information (e.g. what you were trying to achieve)
* Simplest possible steps to reproduce
* Anything that might be relevant in your opinion, such as:
  * MacOS
  * Network configuration
  
### Example

```
Arduino IDE version: 1.8.13
ESP32 Core Version 1.0.5
OS: Ubuntu 20.04 LTS
Linux xy-Inspiron-3593 5.4.0-66-generic #74-Ubuntu SMP Wed Jan 27 22:54:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Context:
I encountered an endless loop while trying to connect to Local WiFi.

Steps to reproduce:
1. ...
2. ...
3. ...
4. ...
```

### Additional context

Add any other context about the problem here.

khoih-prog, 2022-07-22:
Hi @LiveRock 

> and integrate it in my own program

I suggest that you try the original [**Async_ESP32_FSWebServer** example](https://github.com/khoih-prog/ESPAsync_WiFiManager/tree/master/examples/Async_ESP32_FSWebServer) alone, using regular **ESP32_DEV**, Arduino IDE v1.8.19, ESP32 core v2.0.4 first. Be careful with ESP32 TTGO incompatibility (LittleFS, etc.)

If OK, then moving to PIO, etc.

Only merging with your code after all tests are OK.

If problem persists, post the [**Minimal, Reproducible Example**](https://stackoverflow.com/help/minimal-reproducible-example).

I'm closing the issue now and reopen only when you've proven this is a bug of the library.

Good Luck,

closed 2022-07-22 by khoih-prog

LiveRock, 2022-08-01:
I have a different issue now.
When I call

`webServer.addHandler(&events);`

I get the following errors and the board reboot.

```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x401660eb on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f408:0x3ffb37d0 0x4008f681:0x3ffb37f0 0x401660eb:0x3ffb3810 0x40082a4f:0x3ffb3840 0x40089f1d:0x3ffb3860 0x4000bec7:0x3ffb3880 0x401c6ffd:0x3ffb38a0 0x4012fdb9:0x3ffb38c0 0x401efdf7:0x3ffb38e0 0x4013571f:0x3ffb3900 0x40135a3a:0x3ffb3920 0x40135ad6:0x3ffb3960 0x400dd325:0x3ffb3980 0x400ddf1d:0x3ffb3c60 0x400de7b9:0x3ffb3cd0 0x4014ab86:0x3ffb3fb0 0x40090686:0x3ffb3fd0

Rebooting...
```

This is after WIFI is connected and I try to create a webserver to serve JSON data