khoih-prog/ESPAsync_WiFiManager #103: Configuration not saved on azdelivery devkit v2

koen-serneels, 2022-08-16:
Hi. I'm running wifi manager on a az-delivery devkit v2. All is working fine: portal comes online, after entering wifi details the esp is connected to the AP. However, on restart the configuration is not saved. On the az-delivery devkit v4 however, saving works using the exact same code base: settings are saved over restarts. To be sure, flash was erased via esp tool and started over several times, but it's always the same: on the v2 it is not saved while it is saved on the v4 (flashing the same image). Also, the ESP preferences library works on v2 and v4 (stores between restarts). Any idea what could be causing this or any pointers on how to debug this?

khoih-prog, 2022-08-17:
> However, on restart the configuration is not saved. On the **az-delivery devkit v4** however, saving works using the exact same code base: settings are saved over restarts

Sorry can't help here. There are too many different boards / versions on the market, with different design flaws, etc. I have no interest and no way to know what's happening, with too little information.

You have to contact the manufacturers to know what's wrong with v2 and changes from v2-to-v4.

Good Luck,

closed 2022-08-17 by khoih-prog

koen-serneels, 2022-08-19:
@khoih-prog I've investigated further, and it appears to be the AP ssid. My ssid is in the form of "a.b.c 2.4ghz". Connection works, but settings are simply not saved. When connecting to an AP with a more normal ssid (like: "abc") there is no issue and everything gets saved automatically.

So, might this indeed be a but that when an ssid contains the "." (most likely) character things are not getting saved?