khoih-prog/ESPAsync_WiFiManager #108: ESPAsyncDNSServer depends on faulty and not maintained library ESPAsyncUDP

hunter-nl, 2022-08-23:
### Describe the bug

Since ESPAsyncDNSServer is included in ESPAsync_WIFIManger, I get compile error about missing user_interface.h. This is included into ESPAsyncUDP library, used by ESPAsyncDNSServer. user_interface.h is missing and do not know how to solve this.
ESPAsyncUDP is a very old library and not maintained for many years.
In platform.io and expressif32 framework the AsyncUDP library is builtin. Why still use ESPAsyncUDP library?

### Expected behavior

When ESPAsync_WIFIManager is included in my project, the dependencies should compile fine also.

### Actual behavior

It does not compile due depending on a very old and not maintained library ESPAsyncUDP.

### Information

* Platform.io: Core 6.1.4 Home 3.4.3
* ESP32 (Espressif 32): 5.1.0

nicolas-rabault, 2022-08-23:
I have something close to what you describe but I don't see any relation with compile error about missing user_interface.h.
I only have `.pio/libdeps/esp32dev/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.hpp:208:10: fatal error: ESPAsyncDNSServer.h: No such file or directory`
But can't find any way to make PIO find it

nicolas-rabault, 2022-08-23:
Ok I got It to get the same error as you I have to remove the `lib_compat_mode = strict` option.
I think ESPAsyncDNSServer have compatibility issue or something.

khoih-prog, 2022-08-23:
Hi all,

I suggest you try to use Arduino IDE first. Then move on to PIO when everything is OK.

PIO is still **not mature enough for complex use cases**, and requires much more knowledge / skills to **manually modify** the settings.

Post for help in PIO Forum if you still have problem.

I'm closing the issue now, because I just compiled and OK here.

[image]

closed 2022-08-23 by khoih-prog

khoih-prog, 2022-08-23:
> In platform.io and expressif32 framework the AsyncUDP library is builtin. Why still use ESPAsyncUDP library?

Your PIO settings is wrong somehow. Check

https://github.com/khoih-prog/ESPAsyncDNSServer/blob/119dd3ce1b639a5314aac219cf5a00d29946ea47/src/ESPAsyncDNSServer.h#L4-L11

```
#ifdef ARDUINO_ARCH_ESP32
  #ifndef Stream_h
    #include <Stream.h>
  #endif
  #include <AsyncUDP.h>
#else
  #include <ESPAsyncUDP.h>
#endif
```

Post for help in PIO Forum if you can't solve.

---

> I have something close to what you describe but I don't see any relation with compile error about missing user_interface.h.
> I only have [...]

https://registry.platformio.org/libraries/devyte/ESPAsyncDNSServer

LiveRock, 2022-08-31:
Getting this error out of the sudden:

ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.hpp:208:31: fatal error: ESPAsyncDNSServer.h: No such file or directory

How happened?

hunter-nl, 2022-08-31:
platformio.ini can be for example
```
[platformio]
description = Some project description (ESP32)
default_envs = release

[env]
platform = espressif32
board = az-delivery-devkit-v4
framework = arduino
board_build.mcu = esp32
board_build.f_cpu = 240000000L
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_build.partitions = partitions_custom.csv
build_flags =
    -D ARDUINO_ARCH_ESP32

[libs]
builtin =
    SPI
    Wire
    WiFi
    LittleFS
ignore =
    LittleFS_esp32
external =
    bblanchon/ArduinoJson@^6.19.3

[env:release]
lib_deps =
    ${libs.builtin}
    ${libs.external}
	ESPAsyncDNSServer=https://github.com/hunter-nl/ESPAsyncDNSServer
	esphome/ESPAsyncWebServer-esphome @ ^3.0.0
	khoih-prog/ESPAsync_WiFiManager @ ^1.13.0
lib_ignore =
	${libs.ignore}
build_flags =
    ${env.build_flags}
	-D LOG_LEVEL=5

[env:debug]
build_type = debug
lib_deps =
    ${env:release.lib_deps}
lib_ignore =
	${env:release.lib_ignore}
build_flags =
    ${env:release.build_flags}
	-D DEBUG
``` 

I've forked ESPAsyncDNSServer for the DEBUG section. Original it is using ```DEBUG_OUTPUT```, but that does not exist. I've changed that into ```Serial```
In my main.cpp file I include the following lines:
```
#include <Arduino.h>
#include <ArduinoJson.h>
#include <FS.h>
#include <LittleFS.h>

#include <WiFi.h>

#include <ESPAsync_WiFiManager.h>
```
Then it is working fine for me.

LiveRock, 2022-09-05:
I am spending days trying to resolve this without success. Cannot build.
Errors:
.pio/libdeps/ttgo-t8-v17/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.hpp:208:10: fatal error: ESPAsyncDNSServer.h: No such file or directory

My ini file:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[global]
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216

[env:ttgo-t8-v17]
platform = espressif32
board = ttgo-t8-v17
framework = arduino
board_build.mcu = esp32
upload_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 57600
upload_speed = 1152000
board_build.partitions = default_16MB.csv
board_build.f_flash = 40000000L
lib_ldf_mode = deep+
lib_compat_mode = strict
lib_deps = 
	TaskScheduler
	AsyncTCP
	arduino-libraries/Arduino_JSON@^0.1.0
	bodmer/JPEGDecoder@^1.8.1
	siara-cc/Sqlite3Esp32@^2.3
	boseji/xxtea-iot-crypt@^2.0.1
	;bblanchon/ArduinoJson@^6.18.0
	boseji/PString-Arduino-lib@^3.0.2
	spacehuhn/SimpleMap@^1.0.0
	powerbroker2/SafeString@^4.1.11
	rlogiacco/CircularBuffer@^1.3.3
	;bblanchon/ArduinoJson@^6.19.2
	powerbroker2/SafeString@^4.1.15
	bodmer/TFT_eSPI@^2.4.41
	suculent/AESLib@^2.2.1
	densaugeo/base64@^1.3.0
	khoih-prog/ESPAsync_WiFiManager@^1.13.0
	;devyte/ESPAsyncDNSServer@^1.0.0
	;me-no-dev/ESPAsyncUDP
    ${libs.external}
	ESPAsyncDNSServer=https://github.com/hunter-nl/ESPAsyncDNSServer
	esphome/ESPAsyncWebServer-esphome @ ^3.0.0
build_flags = 
	-D ARDUINO_ARCH_ESP32
	-D USER_SETUP_LOADED
	-D ILI9341_DRIVER
	-D TFT_WIDTH=240
	-D TFT_HEIGHT=320
	-D TFT_CS=25
	-D TFT_DC=26
	-D TFT_RST=-1
	-D TFT_MISO=19
	-D TFT_MOSI=23
	-D TFT_SCLK=18
	-D TOUCH_CS=32
	-D LOAD_GLCD
	-D LOAD_FONT2
	-D LOAD_FONT4
	-D LOAD_FONT6
	-D LOAD_FONT7
	-D LOAD_FONT8
	-D LOAD_GFXFF
	-D SMOOTH_FONT
	-D SPI_FREQUENCY=40000000
	-D SPI_READ_FREQUENCY=20000000
	-D SPI_TOUCH_FREQUENCY=2500000
	-D ESP32=1
	-D USE_HSPI_PORT
	-D SUPPORT_TRANSACTIONS
	-D BOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-D CORE_DEBUG_LEVEL=4
	-D CONFIG_SPIRAM_USE_MALLOC=y
	-D CONFIG_SPIRAM_IGNORE_NOTFOUND=y
	-D CONFIG_SPIRAM_MEMTEST=y
	-D CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
	-D CONFIG_SPIRAM_SIZE=4194304
	-D CONFIG_SPIRAM_SPEED_40M=y
	-D CONFIG_NEWLIB_NANO_FORMAT=y
	-D CONFIG_SPI_FLASH_BYPASS_BLOCK_ERASE=y
	-D CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y
	-D CONFIG_SPI_FLASH_ERASE_YIELD_DURATION_MS=40
	-D CONFIG_ARDUINO_LOOP_STACK_SIZE=24576

[env:custom_monitor_port]
monitor_port = /dev/cu.SLAB_USBtoUART
lib_deps = 
	;khoih-prog/ESPAsync_WiFiManager@^1.13.0
	;devyte/ESPAsyncDNSServer@^1.0.0

[env:extra_monitor_filters]
monitor_flags = colorize, b=57600
lib_deps = 
	;khoih-prog/ESPAsync_WiFiManager@^1.13.0
	;devyte/ESPAsyncDNSServer@^1.0.0
[libs]
external =
    bblanchon/ArduinoJson@^6.19.3

[platformio]
description = ESP-WIFI

```

khoih-prog, 2022-09-05:
HI @LiveRock 

> I suggest you try to use Arduino IDE first. Then move on to PIO when everything is OK.
> 
> PIO is still not mature enough for complex use cases, and requires much more knowledge / skills to manually modify the settings.

The PIO already has the published library [**ESPAsyncDNSServer**](https://registry.platformio.org/libraries/devyte/ESPAsyncDNSServer)

It's better to post on PIO Forum for help, if you can't find out how to do.

hunter-nl, 2022-09-05:
@LiveRock the order of the libs in platformio.ini is important. So move ESPAsyncDNSServer before khoih-prog/ESPAsync_WiFiManager@^1.13.0 line
And remove ${libs.external} line as you do not have this section in your platformio.ini file

LiveRock, 2022-09-06:
I re-arranged the sequence as below:

```
lib_deps = 
	devyte/ESPAsyncDNSServer@^1.0.0
	TaskScheduler
	AsyncTCP
	arduino-libraries/Arduino_JSON@^0.1.0
	bodmer/JPEGDecoder@^1.8.1
	siara-cc/Sqlite3Esp32@^2.3
	boseji/xxtea-iot-crypt@^2.0.1
	;bblanchon/ArduinoJson@^6.18.0
	boseji/PString-Arduino-lib@^3.0.2
	spacehuhn/SimpleMap@^1.0.0
	powerbroker2/SafeString@^4.1.11
	rlogiacco/CircularBuffer@^1.3.3
	bblanchon/ArduinoJson@^6.19.2
	powerbroker2/SafeString@^4.1.15
	bodmer/TFT_eSPI@^2.4.41
	suculent/AESLib@^2.2.1
	densaugeo/base64@^1.3.0
	khoih-prog/ESPAsync_WiFiManager@^1.13.0
```

Still getting errors after I CLEAN ALL and BUILD.

Error:
```
Compiling .pio/build/ttgo-t8-v17/lib2e4/Arduino_JSON/cjson/cJSON.c.o
Compiling .pio/build/ttgo-t8-v17/lib864/FS/FS.cpp.o
In file included from .pio/libdeps/ttgo-t8-v17/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.h:37,
                 from src/wificodes.h:174,
                 from src/main.cpp:70:
.pio/libdeps/ttgo-t8-v17/ESPAsync_WiFiManager/src/ESPAsync_WiFiManager.hpp:208:10: fatal error: ESPAsyncDNSServer.h: No such file or directory

***************************************************************************
* Looking for ESPAsyncDNSServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncDNSServer.h"
* Web  > https://registry.platformio.org/search?q=header:ESPAsyncDNSServer.h
*
***************************************************************************

 #include <ESPAsyncDNSServer.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/ttgo-t8-v17/lib864/FS/vfs_api.cpp.o
*** [.pio/build/ttgo-t8-v17/src/main.cpp.o] Error 1
```

LiveRock, 2022-09-06:
Update:
1) I removed lib_compat_mode = strict
2) Added 
```
#ifdef ARDUINO_ARCH_ESP32
#ifndef Stream_h
#include <Stream.h>
#endif
#include <AsyncUDP.h>
#else
#include <ESPAsyncUDP.h>
#endif
```

to ESPAsyncDNSServer.h

Am getting this now:

```
In file included from src/main.cpp:70:
src/wificodes.h: In function 'void doWifiCaptivePortal(String, String)':
src/wificodes.h:973:5: error: 'DNSServer' was not declared in this scope
     DNSServer dnsServer;
     ^~~~~~~~~
src/wificodes.h:973:5: note: suggested alternative: 'webServer'
     DNSServer dnsServer;
     ^~~~~~~~~
     webServer
src/wificodes.h:976:60: error: 'dnsServer' was not declared in this scope
     ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "VT_WIFI_Config");
                                                            ^~~~~~~~~
src/wificodes.h:976:60: note: suggested alternative: 'webServer'
     ESPAsync_WiFiManager ESPAsync_wifiManager(&webServer, &dnsServer, "VT_WIFI_Config");
                                                            ^~~~~~~~~
                                                            webServer
```

khoih-prog, 2022-09-06:
It seems that you still use **old broken** code, before library updated to use `AsyncDNSServer`

Replace any references to `DNSServer` with `AsyncDNSServer`

LiveRock, 2022-09-06:
I didn't go back to Arduino IDE because the project was already working before and suddenly it wasn't. And I use Visual Studio Code daily; so it is preferred. Let me try to replace references from DNSServer to AsyncDNSServer and report back.
Many thanks!

LiveRock, 2022-09-06:
The DNSServer error is gone but I have a whole lot of similar errors as below:

```
In file included from src/main.cpp:152:
/Users/peterlum/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/lib/gcc/xtensa-esp32-elf/8.4.0/include/stdatomic.h:40:9: error: '_Atomic' does not name a type
 typedef _Atomic _Bool atomic_bool;
```

I think it has nothing to do with AsyncDNSServer or ESPAsync_WiFiManager but if someone (khoih-prog?) could help me with this, I would be very very grateful :-)

LiveRock, 2022-09-06:
No more errors after doing

`platform = espressif32@3.5.0`

Thanks!!

rucko24, 2022-11-22:
> It seems that you still use **old broken** code, before library updated to use `AsyncDNSServer`
> 
> Replace any references to `DNSServer` with `AsyncDNSServer`
> 
> Why don't you go back to Arduino IDE to be sure all is OK with your code?

**hahahaha LMAO,** I had been trying for a long time but now it has worked.

Can you update your blog code? 

Really long time, I thought it was all wrong, but it's not. 

At the end it was `AsyncDNSServer`.

thank you.