khoih-prog/ESPAsync_WiFiManager #112: ESP32C3 doesnt show any SSID scan results.

yiancar, 2022-09-04:
### Describe the bug

I am using the Async_ConfigOnDoubleReset_Multi example on an ESP32C3.
I have applied the patches etc. I can go into the config portal but I cannot see any scanned SSIDs.
/scan page shows an empty array.

### Steps to Reproduce

Patch the files for C3
Use the example.

### Expected behavior

Scan page should not be empty.

### Actual behavior
Scan page shows empty variable.

### Debug and AT-command log (if applicable)

`load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000

Starting Async_ConfigOnDoubleReset_Multi using LittleFS on Espressif ESP32-C3-DevKitM-1
ESPAsync_WiFiManager v1.13.0
ESP_DoubleResetDetector v1.3.1
[WM] RFC925 Hostname = AsyncConfigOnDoubleReset
[WM] setSTAStaticIPConfig
ESP Self-Stored: SSID = , Pass =
[WM] LoadWiFiCfgFile
[WM] failed
Open Config Portal without Timeout: No stored Credentials.
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
Starting configuration portal @ 192.168.4.1, SSID = ESP_D0A1DF7C, PWD = MyESP_D0A1DF7C
[WM]
Configuring AP SSID = ESP_D0A1DF7C
[WM] AP PWD = MyESP_D0A1DF7C
[WM] AP Channel = 4
[WM] AP IP address = 192.168.4.1
[WM] HTTP server started
[WM] startConfigPortal : Enter loop
[WM] Handle root
[WM] request host IP = 192.168.4.1
[WM] Handle root
[WM] request host IP = 192.168.4.1
[WM] request host IP = 192.168.4.1
[WM] Handle WiFi
[WM] Static IP = 192.168.2.232
[WM] Sent config page
[WM] request host IP = 192.168.4.1
[WM] Handle WiFi
[WM] Static IP = 192.168.2.232
[WM] Sent config page
[WM] request host IP = 192.168.4.1
[WM] request host IP = 192.168.4.1
[WM] Info
[WM] Info page sent
[WM] request host IP = 192.168.4.1
[WM] Scan
[WM] Scan-Json
[WM] Sent WiFiScan Data in Json format`

### Information

Please ensure to specify the following

-PlatformIO
 - framework-arduinoespressif32 @ 3.20004.220825 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - ESP32 C3

khoih-prog, 2022-09-04:
Hi @yiancar 

I've tested and everything is **OK** here. Be sure to use **Arduino IDE** to first familiarize yourself and to see if your board is OK.

Then moving on to PIO after all is well in Arduino IDE and **you know what you're doing**. Post question / issue next time in PIO Forum if the issue is only for PIO.

I'm closing the issue now.

---

### Config Portal when DRD

```
Starting Async_ConfigOnDoubleReset_Multi using LittleFS on ESP32C3_DEV
ESPAsync_WiFiManager v1.13.0
ESP_DoubleResetDetector v1.3.1
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D01234
doubleResetDetected
Saving config file...
Saving config file OK
Open Config Portal without Timeout: Double Reset Detected
Starting configuration portal @ 192.168.4.1, SSID = ESP_DAA1DF7C, PWD = MyESP_DAA1DF7C
[WM] Connecting to new AP            <======  Save
WiFi connected...yeey :)
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] * Add SSID =  HueNet2 , PW =  12345678
[WM] SaveWiFiCfgFile 
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
[WM] OK
After waiting 0.00 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
HHHH
```

**The WiFi scanning is working and the SSIDs are shown**

[image]

---

### Working normally

```
Starting Async_ConfigOnDoubleReset_Multi using LittleFS on ESP32C3_DEV
ESPAsync_WiFiManager v1.13.0
ESP_DoubleResetDetector v1.3.1
ESP Self-Stored: SSID = HueNet1, Pass = 12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
Got ESP Self-Stored Credentials. Timeout 120s for Config Portal
[WM] LoadWiFiCfgFile 
[WM] OK
[WM] stationIP = 192.168.2.232 , gatewayIP = 192.168.2.1
[WM] netMask = 255.255.255.0
[WM] dns1IP = 192.168.2.1 , dns2IP = 8.8.8.8
Got stored Credentials. Timeout 120s for Config Portal
LittleFS Flag read = 0xD0D04321
No doubleResetDetected
Saving config file...
Saving config file OK
ConnectMultiWiFi in setup
[WM] ConnectMultiWiFi with :
[WM] * Flash-stored Router_SSID =  HueNet1 , Router_Pass =  12345678
[WM] * Add SSID =  HueNet1 , PW =  12345678
[WM] Connecting MultiWifi...
[WM] WiFi connected after time:  1
[WM] SSID: HueNet1 ,RSSI= -24
[WM] Channel: 4 ,IP address: 192.168.2.232
After waiting 5.87 secs more in setup(), connection result is connected. Local IP: 192.168.2.232
HStop doubleResetDetecting
Saving config file...
Saving config file OK

```

closed 2022-09-04 by khoih-prog

yiancar, 2022-09-04:
Hello @khoih-prog thank you for taking a look. Is there a chance you can also test with PIO as well?
I am trialing Arduino IDE right now

khoih-prog, 2022-09-04:
No, I don't have time to test and also am not interested in using PIO.

Good Luck,

yiancar, 2022-09-04:
Hello:) I have tried with the arduino IDE and the issue persists.
If i open configportal and manually type the ssid/pass the module connects succesfully.
If you have any other debug tips before I order another module please LMK

khoih-prog, 2022-09-04:
It seems there is some problem with your board (**bad, fake**, etc.) and only you can figure out by testing.

Try to modify

https://github.com/khoih-prog/ESPAsync_WiFiManager/blob/13593167119d155cb83492f3b3bb10209b9bcecf/examples/Async_ConfigOnDoubleReset_Multi/Async_ConfigOnDoubleReset_Multi.h#L48-L49

to

```
// Use from 0 to 4. Higher number, more debugging messages and memory usage.
#define _ESPASYNC_WIFIMGR_LOGLEVEL_    4
```

then check to see what's happening.

It's better to order another board or boards from reliable / respected suppliers.

Post on Arduino or ESP32 Forum to see if someone got the same ESP32_C3 issue (**WiFi, LittleFS, etc**).