khoih-prog/ESPAsync_WiFiManager #113: "no protocol" error

ShadowSA, 2022-09-15:
### Describe the bug

When using the library manager in Arduino IDE, searching for ESPAsync_WiFiManager, then clicking install, throws error: "no protocol"

### Steps to Reproduce

Open Arduino IDE, select library manager, search ESPAsync_WiFiManager, click install on version 1.14.0

```

khoih-prog, 2022-09-15:
You have to select `Install ESPAsync_WiFiManager only` to install the library.

The reason is that it depends on some other libraries which are **not** `published` or `well maintained` on Arduino IDE. You have to install them manually by downloading `zip`.

I'll possibly remove the dependency soon and make some notes about the issue.

Research or ask on Arduino Forum if you don't know how to install a library manually.

- ESP Async WebServer for both ESP32/ESP8266
- ESP AsyncTCP for both ESP8266
- AsyncTCP for both ESP32

[image]

[image]

closed 2022-09-15 by khoih-prog

khoih-prog, 2022-09-15:
Hi @ShadowSA 

The [**ESPAsync_WiFiManager** v1.14.1](https://github.com/khoih-prog/ESPAsync_WiFiManager/releases/tag/v1.14.1) has just been released to address the `dependency` issue.

Please test.

Regards,

---

### Releases v1.14.1

1. Remove dependency on ESP_AsyncWebServer, ESPAsyncTCP and AsyncTCP in `library.properties`. Check ["no protocol" error #113](https://github.com/khoih-prog/ESPAsync_WiFiManager/issues/113)"
2. Add notes about manual installation of above-mentioned libraries for `Arduino IDE`